v002/v002-Camera-Live #5: Camera Live in El Capitan

dimbels, 2016-01-13:
Did anyone have success with running Camera Live in El Capitain? it does not seem to be working for me. Canon2Syphon works, but i would like to use two cameras.
Also updated the Eos Utility because that did not seem to run on el capitain anymore. 
Maybe a general matter of SDK incompatibility?

glidaa, 2016-03-06:
Im having the same issue, is there any way to get this camera live rebuilt, it needs the cannon sdk and they dont seem to give that out without a 5 -10 day review

bangnoise, 2016-03-07:
waiting 5 to 10 days for SDK access would be one solution.

glidaa, 2016-03-07:
i just tried to rebuild with another sdk, wow the errors are a bit beyond my skill, is there anyone who can pickup the awersome app?

bangnoise, 2016-03-08:
I'm looking at the current SDK's stability - I may do an updated release in
the next couple of weeks.

glidaa, 2016-03-08:
greatnews

thomasvanta, 2016-04-22:
Just tried with EDSDK 3.4, last libjpeg-turbo as of today under El Capitan and is working fine, out of the box.
:)

glidaa, 2016-04-22:
Can you give a link to your build

bangnoise, 2016-04-22:
I've just posted [v5](https://github.com/v002/v002-Camera-Live/releases/tag/v5) with the new Canon SDK.

If you run into problems with it, please open a new issue.

closed 2016-04-22 by bangnoise