v002/v002-Camera-Live #8: App is broken on MacOS 10.12 (Sierra)

beamtown, 2016-10-28:
Recently updated to MacOS 10.12 and the app no longer functions. The problem manifests in two ways, depending on launch order:

1. If the camera is connected to the computer via USB and turned on before Camera Live is opened, the app will not launch and just continually bounces in the dock.

2. If the app is opened before connecting the camera to the computer, it will launch successfully. However, after connecting the camera and turning it on, Camera Live will not recognize the input.

For what it's worth: HUGE thanks for making this app. It's been incredibly useful to be able to use my canon SLR as a web cam without needing a capture card. I really hope it can be updated sometime soon, but understand that bug fixing is a slow process, especially for amazing freeware like this. Just wanted to pass along my gratitude for your work, regardless of whether or not this issue is fixed!

bangnoise, 2016-11-01:
Still an issue in latest Canon SDK (3.5).
Hanging in `EdsOpenSession()`, which works in Canon sample code, perhaps SDK has issue with a dispatch queue (sample code uses a single thread).

wmwood, 2016-12-09:
Not working for me on macOS 10.12.1

roolo, 2016-12-11:
So the fix i mentioned is working just time to time. Today it was working for a while, but now nothing again :/

mitchdurfee, 2017-01-24:
Hello, I have tried a ton of different things and I have been able to get the camera live to turn on, I have had it work once in a great moon out of 100 times of clicking it on and off but if there is any thing I can do to get this up and running I really some option of getting this running...

andresrivero, 2017-03-02:
This app works fine in Sierra. I use it with OBS. I start Canon EU first, quit, then start Camera.Live. Please check your settings or reinstall your OS.

bangnoise, 2017-03-02:
There is an issue with macOS 10.12 which seems to affect some devices more severely than others. Please no more "me too" posts. Nobody need reinstall their OS.

ben-walters, 2017-03-02:
At present, I am able to start the application on 10.12.3, but displays a "starting" message, and never completes. Im not sure if this is similar to anyone else.

grizlie, 2017-03-06:
Signed on just to share this.  I've managed to get it working, and have tested it numerous times to ensure it's not random.

As a user shared earlier, here's the process I follow:
1. Connect Canon 600D DSLR via USB and switch into Video mode
2. Turn on EOS Utility (do not go into Live View)
3. Open Camera Live (window will show your camera and in 'Ready' status - do not click on the Camera Live app)
4. Exit EOS Utility (at this point, Camera Live still shows 'Ready')
5. In the Camera Live window, select your camera (in my case, I select Canon EOS 600D)

When I select my camera, the status changes from Ready>Starting>Active.  The camera responds accordingly by 'reconnecting'.  From here, you launch CamTwist and follow the process from there.

I'm thrilled this is working - two day after I bought an Elgato Game Capture HD for the purposes of capturing footage to my laptop!

Hope this works for you guys.  Blessings.

brendanrbercik, 2017-03-24:
Cool, I am installing EOS Utility now. Any chance there will be an update to Camera Live that doesn't require this workaround? When I first installed it, I did not have EOS Utility on my computer. Would be great if it could be fully standalone again.

gregclermont, 2017-07-02:
If anyone is interested in a slightly simpler workaround, I found out that opening Photos.app instead of EOS Utility works too.

Also, I think that the bug can be prevented if you always deselect the camera in Camera Live before closing it. (Not sure, I don't have a camera to reproduce right now).

brendanrbercik, 2017-07-08:
Thanks for the tips grizlie. Tried your method, and much like ben-walters I cannot get the camera past "Starting." Camera Live appears as a Syphon Server in Syphon Apps (such as Lumen, Simple Client, Arena) but does not produce video.

I hear the mirror flip up when I select my camera in Camera Live (step 5) but never reach "Active."

macOS Sierra 10.12.5
Macbook Pro 15" Late 2015
AMD Radeon r9 M3790X

L-Kov, 2017-07-13:
@ben-walters did you get it figured out? Mine still doesn't seem to work.

L-Kov, 2017-07-13:
@brendanrbercik did you get it figured out? Mine still doesn't seem to work.

brendanrbercik, 2017-07-13:
I did not. I know we don't want multiple "me too" posts, but if you provide the most detailed information of the issue that you can, it may help speed up the devs' process.

bangnoise, 2017-07-14:
This bug is easily reproduced. I don't need any more information, thanks - nor people repeating that it doesn't work.

MarkCACasiano, 2017-07-15:
Here is how I mad it work, start Canon EOS utility, start Camera Live, make sure camera is listed but not selected. If camera is selected, unselect it and quit Canon EOS utility including from the dock, and quit camera live. then open Canon EOS utilty, Open Camera Live, quit all Canon EOS utility including from the dock, then select your camera from Camera Live. Now it worked on my end. Let me know if it does to you.

closed 2017-08-15 by bangnoise

bangnoise, 2017-08-15:
Try the [latest build](https://github.com/v002/v002-Camera-Live/releases).

commit 1f33caa3 (2017-08-15): Throttle calls to EdsOpenSession

Together with 09acf9bfd1470f745265cd122471ace25a17c795 this fixes #8