v002/v002-Camera-Live #10: maybe consider libgphoto

bangnoise, 2017-04-03:
Haven't looked into its suitability, but maybe consider ditching the Canon SDK and using [libgphoto](http://www.gphoto.org).

bangnoise, 2020-03-24:
see libgphoto2 branch

deanchen, 2020-03-27:
libgphoto2 build has been working great on both my Canon 77D and Sony a7riv!

bangnoise, 2020-03-27:
@deanchen ah thanks for the confirmation - glad to hear it's working on the Sony especially.

ikester, 2020-05-13:
> see libgphoto2 branch

Thanks for this utility. Very helpful, especially in these times.

Do you have any recommendations for building the libgphoto2 branch? I'm struggling with getting all the libraries in place. I used brew to download many of the dependencies, including jpeg-turbo, but XCode is now complaining about missing lusb-1.0.0.

bangnoise, 2020-05-18:
@ikester I'll provide step by step build instructions for the required libraries once this branch is a bit more stable - I'm building libltdl, libusb and libgphoto2 from source and using install_name_tool to change their link IDs post-build to allow linking from the app bundle - it's a bit of a pain. Versions from Homebrew should work - if you're only building for use on your own system you could modify the project to link to the Homebrew versions, otherwise you copy them and modify them as above, so they aren't linking to the Homebrew-installed versions. `otool -L` and `install_name_tool --` are your friends. You will also have to do the same for the plugins libgphoto2 uses.