v002/v002-Camera-Live #11: Camera Error message

ocervantes18, 2017-04-26:
Hello,

I'm trying to connect a Cannon 5D Mark IV camera and once I launch Camera Live, turn on the camera and select the device, I get a "Camera Error" message and it won't go live. Any help?

HillAndrewP, 2017-06-09:
I too am using a 5D Mark IV, and have a feeling it's simply unsupported? Excited for it to hopefully be covered soon--

Thanks for the awesome app and support for all previous cameras.
[image]

lsapan, 2017-07-03:
@bangnoise thanks for creating this, it works great with other cameras! I know nothing about the Canon SDK or where to get it, but if there are instructions anywhere I'd be happy to try getting the latest SDK up and running to add 5D Mark IV support. Do you have any pointers?

vade, 2017-07-05:
Updated the issue title for clarification.

bangnoise, 2017-07-05:
Not really worth putting out an update with the new SDK unless the new SDK also fixes #8

HillAndrewP, 2017-07-05:
With the most recent Canon software and macOS Sierra 10.12.5, everything Canon + Sierra side works perfectly for me and my 5D Mark IV with a Mid-2015 MacBook Pro (Graphics: Intel Iris Pro 1536 MB).  The only missing piece is Camera Live supporting the 5D Mark IV.

Which is just to say, putting out an update with the new SDK for 5D Mark IV support would indeed be helpful to at least me, and presumably the other people in #8 and #11 who aren't experiencing any issues with the Canon + Sierra software.

bangnoise, 2017-08-15:
Should work now with the [latest release](https://github.com/v002/v002-Camera-Live/releases). Re-open this issue if not.

closed 2017-08-15 by bangnoise

HillAndrewP, 2017-08-16:
I'm still finding the same issue, though if anyone has special order of operations, please let me know.  The list below is my procedure, and typically finds success with pre-5D Mark IV cameras:

1. Downloaded the newest 'Camera Live' (Thanks @bangnoise )
2. Disabled and exited all Canon Software
3. Plugged Camera into USB 3.0
4. Turn Camera On (Idle and Busy for USB connection)
5.1 Open newest Camera Live
5.2 Camera display changes from black, shows computer, recognizes computer connection
5.3 Camera briefly flashes red light for record (though EOS Utility stays red when successfully connected)
5.4 Camera Live status briefly switches to 'Starting' before sticking on 'Camera Error' for Canon EOS 5D Mark IV

Any advice or confirmation if it's working for anyone else?

GameBraining, 2020-05-14:
Hi all, anyone my canon eos m10 is having issues connecting
[image]