v002/v002-Camera-Live #15: Not working for Canon 1300D

CanonUser, 2017-07-15:


bangnoise, 2017-08-15:
This should have been working when you opened this issue. Can you try the [latest build](https://github.com/v002/v002-Camera-Live/releases) and if the issue persists, provide more detail (eg any errors, or a screenshot of the window with the camera connected).

Basdhen, 2017-09-14:
It's not recognizing my Canon 1300D either.
Mac 10.12.6
Canon 1300D firmware version 1.1.0
Tried it with build #6

Canon 70D working perfectly fine.

bangnoise, 2017-09-14:
Does the camera show up in Camera Live at all, or is completely ignored?

Basdhen, 2017-09-14:
The camera is getting completely ignored.

bangnoise, 2017-09-14:
Can you confirm you don't have EOS Utility (or any other software which might be using the camera) open?

Basdhen, 2017-09-14:
I do have EOS Utility but the problem was that the WIFI/NFC function was enabled.
It works now. Sorry for the inconvenience

bangnoise, 2017-09-14:
No problem - useful information thanks, hopefully also the solution for @CanonUser .

closed 2017-09-14 by bangnoise