v002/v002-Camera-Live #17: "Camera Error" message when connecting Canon 5D Mark IV

HillAndrewP, 2017-08-16:
Hi, the previous issue was closed, but @bangnoise said to open it again if the issue persists.

Feel free to open the previous issue (#11) or follow up here, because I wasn't able to re-open it myself.  Copying my post from the previous thread below:

> I'm still finding the same issue, though if anyone has special order of operations, please let me know. The list below is my procedure, and typically finds success with pre-5D Mark IV cameras:

> 1.Downloaded the newest 'Camera Live' (Thanks @bangnoise )
> 2. Disabled and exited all Canon Software
> 3. Plugged Camera into USB 3.0
> 4. Turn Camera On (Idle and Busy for USB connection)
> 5.1 Open newest Camera Live
> 5.2 Camera display changes from black, shows computer, recognizes computer connection
> 5.3 Camera briefly flashes red light for record (though EOS Utility stays red when successfully connected)
> 5.4 Camera Live status briefly switches to 'Starting' before sticking on 'Camera Error' for Canon EOS 5D Mark IV

> Any advice or confirmation if it's working for anyone else?

brendanrbercik, 2017-08-16:
Is there a firmware update available for your camera?

If any other 5D Mk IV users report a success, can you please post firmware version?

HillAndrewP, 2017-08-16:
Perfect catch, @brendanrbercik!

I updated to the latest firmware (1.0.4), and now have full support.  Thank you!

For other and future Canon 5D Mark IV users, download the latest firmware here.

https://www.usa.canon.com/internet/portal/us/home/support/details/cameras/dslr/eos-5d-mark-iv?subtab=downloads-firmware

closed 2017-08-16 by HillAndrewP