v002/v002-Camera-Live #18: Not recognizing camera

amselina, 2017-08-17:
Hello,
I have an SL1 and it is not recognizing my camera anymore.  I just upgraded to the new build, so that's not the problem.  It pops up like it's going to recognize it and then proceeds to say camera error and then no camera.

bangnoise, 2017-08-17:
When you say anymore, did it work with a previous build of Camera Live? If so, which build?

bangnoise, 2017-08-19:
Closing this as duplicate of #16 - please provide more information there, thanks.

closed 2017-08-19 by bangnoise

amselina, 2017-08-19:
I mean it's just saying starting starting starting and then shuts down.
This is the most recent build, which was your solution to 16.

bangnoise, 2017-08-19:
Did it ever work with any previous build of Camera Live?

amselina, 2017-08-20:
It worked well with the last one.

Sent from my T.A.R.D.I.S

bangnoise, 2017-08-20:
Thanks - what OS version are you on?

Unfortunately I don't have access to this camera to test with. In the meantime you can use the previous version.