v002/v002-Camera-Live #19: MacOS High Sierra Crashing

ejwhite7, 2017-08-31:
When I open CL and connect my Canon t3 (which works with previous OSs) CL crashes immediately. Thanks!

bangnoise, 2017-08-31:
You sure you're on the [latest release](https://github.com/v002/v002-Camera-Live/releases)?

Please paste the crash report here if so.

bangnoise, 2017-09-04:
I'm still occasionally seeing crashes on EdsOpenSession() with my 7D, looks like we need to try to avoid the open/close/open session cycle entirely.

kevindiaz, 2017-09-06:
Just wanted to confirm that I am having the same issue on High Sierra V9. The app crashes the moment I hear the shutter open. The app says "Active" briefly, the crashes. 5D MK3

matkomatkovic, 2017-09-26:
I can also confirm that app is crushing when ran on High Sierra OS. I am running latest version 6 and having Canon Rebel T2i connected.

[HighSierra.txt](https://github.com/v002/v002-Camera-Live/files/1335059/HighSierra.txt)

kc9m92017, 2017-09-27:
I'm using the Canon T3i. The Program was working fine until upgrade to MacOS High Sierra, now every time i open it shutter open program connects shutter clicks again and program closes. See attached error report.

Really good program, best i have found to get the job done for mac. Hope a fix will be available soon.
Crash Report Attached:
[HighSierra.txt](https://github.com/v002/v002-Camera-Live/files/1337876/HighSierra.txt)

Thanks,

bangnoise, 2017-09-27:
No more "me-too" posts please

TomTomTomGit, 2017-10-07:
Where you already able to work on this bug?

If it helps, I could provide a crash log too.

ricardoclarke, 2017-10-18:
Hey @bangnoise If I contributed some funds, would that help to work on this bug? It's such a cool app and I hate that I can't use it anymore because of an Apple update.

DavidSpriggs, 2017-10-19:
@ricardoclarke, @bangnoise Same here, I can contribute as well.

bangnoise, 2017-11-02:
Could anyone try again on 10.13.1? I can't reproduce this with a 7D.

ricardoclarke, 2017-11-02:
I've upgraded to 10.13.1 and I am still getting the same error with my 70D. I've attached the error log. 

[CameraLive Error Log.txt](https://github.com/v002/v002-Camera-Live/files/1439163/CameraLive.Error.Log.txt)

bangnoise, 2017-11-03:
Looks like (yet another) bug in the Canon SDK. Could you try deleting any images stored on the camera (or inserting blank media) and then try again?

ricardoclarke, 2017-11-03:
That actually worked! Strange, but popped in an empty SD card and the app was working like normal. I then put in my normal card and Camera Live crashed and the camera turned off. 

Thank you. I'll take it :)

kaspesla, 2017-11-05:
Hi Tom, I have a fix for this, which is to just swizzle out the ImageCaptureCore method that's causing the photos directory parsing that's apparently causing this to die. I could do a pull request or just email you the code?

bangnoise, 2017-11-05:
@kaspesla sounds brutal but sure - PR or mail welcome

kaspesla, 2017-11-05:
Not as bad as it sounds. :) Sending PR
-Ken

ThomasBallatore, 2017-11-08:
@bangnoise Popping the card out of my 5D Mark II worked after an upgrade to High Sierra crippled things. Thanks *so* much for this app!

bangnoise, 2017-11-08:
Thanks to @kaspesla there's a new [release](https://github.com/v002/v002-Camera-Live/releases) up now which fixes this issue.

closed 2017-11-08 by bangnoise