v002/v002-Camera-Live #22: "Camera Error" Canon 80D

kgeary83, 2017-09-13:
All I get is a "camera error" when trying to hook up the Canon 80D...

bangnoise, 2017-09-13:
What version of macOS? What version of Camera Live?

kgeary83, 2017-09-13:
MacOs: 10.12.6
CameraLive: Version 6

bangnoise, 2017-09-13:
Some cameras require that they are manually put in movie or live-view mode, you could try that.

kgeary83, 2017-09-13:
I've tried switching it into live view and movie mode manually. It doesn't work. I was successful in getting the camera connected with EOS Utility. But when I close EOS Utility and try camera live, it just says error.

bangnoise, 2017-09-13:
Do you hear the shutter open or are there any other indications of activity from the camera before it reports the error?

kgeary83, 2017-09-13:
The shutter clunks and then the LCD on the camera displays a monitor symbol.

bangnoise, 2017-09-13:
Strange. Okay I'm afraid out of ideas.

o2eric, 2017-10-04:
I had the same, as workaround I used camtwiststudio.com it filled missing s/w relations.
Than Camera Live became ACTIVE  and it works fine now. 
Can you confirm this work for you too.

kgeary83, 2017-10-04:
​Nope, doesn't seem to work.​

*Sent with Shift
<https://tryshift.com/?utm_source=SentWithShift&utm_campaign=Sent%20with%20Shift%20Signature&utm_medium=Email%20Signature&utm_content=General%20Email%20Group>*

justin271190, 2017-10-07:
Even I am getting the same issue on 80D. Also tried solution given by @o2eric but that didn't help.  Does anyone else have any other workaround?

cumulativerecords, 2017-10-09:
I have v6 CameraLive working with 80D (firmware 1.0.2) on macOS 10.12.6 Sierra. Did not run EOS utility. 

When I connect 80D usb and open CameraLive:
QuickTime does not recognize the 80D as a camera, but then I opened obs-studio and it did recognize the 80D as a "Game Capture" device.

donadisal, 2017-12-06:
I have canon 700D, after upgrade to High Sierra 10.13.1 camlive is not picking anymore
I tried with Yosemite and its working. I don't know if its the cannon EOS utility.

maksad, 2018-02-28:
I had the same issue with 
* EOS 550D
* MacOS HighSierra 10.13.3
* Camlive: Build 8
* Cam twist: Version 3.4 (3.4)

It did not work at first, but after reading @cumulativerecords comments `Did not run EOS utility.` I closed EOS utility and it started working

aeickhoff, 2018-03-19:
For me it started working after updating camera firmware (v101 -> v102)

lyrium, 2018-06-28:
Necro'ing this — Canon 80D still doesn't work "Camera Error".

- MacOS Sierra 10.12.6 
- Camera Live v9
- Camtwist 3.4

TheBestme09, 2018-07-04:
I am stilling get the same message as well. Any suggestions. It did work last week but now it will not connect.

luisramalho, 2018-07-05:
@TheBestme09 @lyrium I had the same issue just now and upgrading the firmware worked for me as aeickhoff pointed out EOS 80D v101 -> v102

luisramalho, 2018-07-07:
yes

Are you using a macbook?

*Jahbari McLennan*
Ziglar Legacy Certified Trainer
*"You are where you are and what you are because of what’s gone into your
mind; you can change where you are and what you are by changing what goes
into your mind"*
www.jahbarimclennan.com<http://www.jahbarimclennan.com>

TheBestme09, 2018-07-12:
@luisramalho you are right it did work! Thank you!

jakebam4th, 2019-03-31:
I get the same Camera Error Message

Running mac Mojave 10.14.3
Canon t3i 600d
Camera live Version 9

I've also got Camtwist Version 3.4.2

No luck.. Anyone have an idea?

bangnoise, 2019-11-19:
Please try with [the latest release](https://github.com/v002/v002-Camera-Live/releases/latest).

closed 2019-12-02 by bangnoise