v002/v002-Camera-Live #24: Multiple Camera feeds

JonDRichardson, 2017-10-10:
This is less of an issue and more of a general question. I'm trying to connect two DSLR cameras through OBS for live streaming. I tried connecting two 5D Mark IIIs to Camera Live (opened two instances of it via Terminal in OS X Sierra), but it only seems to ever want to have one at a time. I also tried it with one 5D and one 7D. Anyone have any guidance? Is this even possible?

bangnoise, 2017-10-12:
Not sure it's possible, sorry.

closed 2017-10-12 by bangnoise

JonDRichardson, 2017-10-12:
I actually managed to find a workaround, at least for my needs. If I run Canon2Syphon, it'll grab one camera and then Camera Live can grab the other. This wouldn't work for folks using non-Canon cameras.