v002/v002-Camera-Live #27: Camera Live crashes when connecting to Canon 70D

brettgoodrich, 2017-11-09:
Running High Sierra 10.13, Camera Live version 7, latest firmware (1.1.2) on 70D.
EOS Utility set to not start automatically.

I start with the camera disconnected and set to video. I open Camera Live, push in the USB. It sees the 70D, and the camera screen shows the little laptop icon. I click the 70D in Camera Live and it says "Starting" for a quarter-second and then the window disappears. About four to five seconds later, I get a classic "Camera Live quit unexpectedly" window, with a report button and everything.

Is the 70D incompatible?

gaspesla, 2017-11-10:
This sounds exactly like the issue fixed in release 7, just yesterday. Do you definitely have the latest release?

brettgoodrich, 2017-11-10:
Yes. It has an icon, so I’m sure.

-- 

*-Brett Goodrich*

*Youth Pastor*

O 317-823-2349

C 317-402-7879

brettgoodrich@horizonindy.org

*Horizon Christian Fellowship*

7702 Indian Lake Road

Indianapolis, Indiana

317-823-2349

www.horizonindy.org

What's Horizon <http://vimeo.com/93440072>?

bangnoise, 2017-11-10:
If you let it crash and then click "Report..." in the dialog that appears, can you copy and paste the crash report (or attach it) please?

brettgoodrich, 2017-11-10:
[CameraLive_70D_crashreport.txt](https://github.com/v002/v002-Camera-Live/files/1461677/CameraLive_70D_crashreport.txt)

kaspesla, 2017-11-10:
Looks like it's ending up in the same crash, just taking a different code path. :( Could probably patch it in a better spot.

bangnoise, 2017-11-10:
@brettgoodrich can you give [this build](https://github.com/v002/v002-Camera-Live/releases/download/v8/Camera.Live.zip) a go and report back?

brettgoodrich, 2017-11-11:
@bangnoise Works perfect! No crash. Thank you!

closed 2017-11-11 by bangnoise

brettgoodrich, 2017-11-14:
Hey @bangnoise, this is more of a support question: what resolution is Syphon giving me off the camera?
I can't figure out a way to easily tell.

bangnoise, 2017-11-14:
Whatever you're sending to can often tell you - you could use [Simple Client](https://github.com/Syphon/Simple/releases/download/version-3/Syphon.Simple.Apps.3.zip).

brettgoodrich, 2017-11-14:
@bangnoise thank you!

brettgoodrich, 2017-11-16:
@bangnoise is there any way to choose what resolution you want out of the camera, framerate be darned, or is it hard-set?

bangnoise, 2017-11-16:
Afraid it's not configurable.

MarkusHeitkoetter, 2018-01-01:
Just FYI: I have the latest release 8 (8) and had the same problem with a Canon Rebel T6. 

I removed the SIM card and everything is working perfectly.
With a SIM card it's hit and miss.

Hope this helps with troubleshooting...