v002/v002-Camera-Live #28: Problem with 5D Mark 3

dragonkiat, 2017-12-03:
Yesterday I first tried Camera Live on my 5D mark3 and Macbook pro with no issue at all . But today when I try again , it's not working . There's no video input to OBS although I've repeat every steps many times . 
My issue is , after start up the Camera Live app and switch on my 5D mark3 in video recording mode , the liveview screen on camera body will shut down and keep asking me to press the START/STOP button to re-activate liveview again . But , the liveview can't be activated again unless I unplug the Canon USB cable .
Do U have any sollution ? Thank you .

bangnoise, 2017-12-03:
I don't have access to this camera, but I'd suggest you try with the camera in photo-taking mode.

Adrian-Tucker, 2018-11-05:
Hi! I connected my 700D and worked perfectly the first time, now I connect it the same way and the camera live app freezes in starting... Any solution?

Cheers!

bangnoise, 2018-11-05:
@Adrian-Tucker as above, check which mode the camera is in. There are also major issues with the Canon SDK and recent macOS versions, and it's possibly you happen to have hit those.

bangnoise, 2019-11-19:
Please comment if this issue persists in [the latest release](https://github.com/v002/v002-Camera-Live/releases/latest).

closed 2019-11-19 by bangnoise