v002/v002-Camera-Live #31: 2 Camera's

Wallymus, 2018-01-16:
Hiya! 

First off; Wonderful app you created!

I came across a problem just now: I wanted to start streaming with 2 camera's (Canon 600D and 6D) from 1 laptop through OBS, but Camera Live doesn't seem to take 2 camera's at the same time. Seperately both camera's work, but when I try them both at the same time the 600D won't connect (it is visible in Camera Live though..)

Hope you can help! Thanks!

bangnoise, 2018-01-16:
There are two factors at play here - the bandwidth of the USB bus and a software limitation imposed by the Canon SDK.

You could try making two copies of the Camera Live and launching them both - that will circumvent the software limitation. There's not much you can do if that still won't work.

Wallymus, 2018-01-16:
I'll try, thanks!

Doormandean, 2018-11-04:
Before Camera Live stopped working completely with Mojave, I ran two canon cameras (77D and 200D) by cloning the app, so I could have two running at the same time. Occasionally one would crash, but generally it was great

bangnoise, 2020-05-21:
Closing this in favour of #76

closed 2020-05-21 by bangnoise