v002/v002-Camera-Live #33: Camera Error - or Status "Starting" EOS 500D

farzadprivate, 2018-03-05:
Trying to get this to work. I don't know what I changed but earlier I saw the status "ready" - and when clicking on the cam it would say Camera Error. I think that was because EOS UTILITY was running in the background.

MacOS 10.13.3 - but I am about to update to see if issue persists...

Thanks already for any help...

farzadprivate, 2018-03-05:
Update: Did even last update of macOS and now even eos utility won't connect to Camera. It says: "Failed to communicate with camera - check connections"

bangnoise, 2018-04-18:
Don't use EOS Utility and Camera Live at the same time. Check the [Troubleshooting](https://github.com/v002/v002-Camera-Live#troubleshooting) section of the Read Me.

closed 2018-04-18 by bangnoise

themufi, 2020-06-03:
Same issue here. my device is actually a rebelt t1i. Cam live picks it up as a 500D. Tired everything on the troubleshooting - no luck. It just fails to move beyond camera error

megerman, 2020-07-06:
Same issue here. I suspect the T1i/500D doesn't actually work on Mac OS - Utility 2 also won't see it.

thooams, 2021-03-09:
Same error with my camera EOS 500D and without EOS Utility.
[image]

How to do to fix that?

Mac: Mac Book pro 2019
Os: Big Sur