v002/v002-Camera-Live #34: Camera Live is Not Optimized for Your Mac

AlfredoBenitez, 2018-04-18:
After the last actualization for the mac, camera live sends the message that is not working.  Cna you pleas tell me what can i do, i am making facebook livestreams with a canon 70d and now does not work!

Thanks in advance!
[image]

bangnoise, 2018-04-18:
What is the message you see?

There are various issues with the Canon SDK and macOS 10.13 which I can't do anything about (short of a complete rewrite without the official Canon SDK).

Have you followed the steps in the Troubleshooting section of the [Readme](https://github.com/v002/v002-Camera-Live#troubleshooting)?

vade, 2018-04-18:
> Have you followed the steps in the Troubleshooting section of the Readme?

bangnoise, 2018-04-18:
The "not optimized for your mac" message is unrelated. The app will still open when you dismiss the message.

closed 2018-04-18 by bangnoise

AlfredoBenitez, 2018-04-18:
thank you. :D It is working