v002/v002-Camera-Live #38: Crash on App Start

Naddi-Squared, 2018-06-27:
Camere Live all versions keep crashing on open.. the window opens for a second and then it crashes...

bangnoise, 2018-06-27:
If you click on "Report..." on the crash dialog, could you copy and paste the crash report that is displayed here?

hwschuur, 2020-04-09:
Here's a crash report from Camera Live build 11 in macOS Catalina 10.15.4:

```Process:               Camera Live [52302]
Path:                  /Applications/Camera Live-official.app/Contents/MacOS/Camera Live
Identifier:            info.v002.Camera-Live
Version:               11 (11)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           Camera Live [52302]
User ID:               502

Date/Time:             2020-04-09 13:53:30.445 +0200
OS Version:            Mac OS X 10.15.4 (19E266)
Report Version:        12
Bridge OS Version:     4.4 (17P4263)
Anonymous UUID:        45AADD1A-57E8-5E43-79A7-271DE37AA695

Sleep/Wake UUID:       24674A42-FDE1-40DE-ACC4-0EC2B520F4DA

Time Awake Since Boot: 170000 seconds
Time Since Wake:       3900 seconds

System Integrity Protection: enabled

Crashed Thread:        1  Dispatch queue: info.v002.Camera-Live.Canon.liveview

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000058
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [52302]

VM Regions Near 0x58:
--> 
    __TEXT                 00000001061b8000-00000001061c3000 [   44K] r-x/r-x SM=COW  /Applications/Camera Live-official.app/Contents/MacOS/Camera Live

Thread 0:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff73c8fdfa mach_msg_trap + 10
1   libsystem_kernel.dylib        	0x00007fff73c90170 mach_msg + 60
2   com.apple.framework.IOKit     	0x00007fff3c9757f3 io_connect_method + 383
3   com.apple.framework.IOKit     	0x00007fff3c97d625 IOConnectCallScalarMethod + 76
4   com.apple.iokit.IOUSBLib      	0x000000010b36dc79 IOUSBDeviceClass::USBDeviceOpen(bool) + 117
5   com.canon.edsdk               	0x00000001062f2bdd darwin_open + 103
6   com.canon.edsdk               	0x00000001062f0c6a libusb_open + 223
7   com.canon.edsdk               	0x00000001062f0d94 libusb_open_device_with_vid_pid + 140
8   com.canon.edsdk               	0x000000010630f8fd CPtpLinuxUSBDevice::OpenDevice() + 35
9   com.canon.edsdk               	0x000000010630f869 CPtpLinuxUSBDevice::SetupConnection(int) + 21
10  com.canon.edsdk               	0x000000010630f6bd CPtpLinuxUSBDevice::AddUSBInfo(libusb_device*) + 343
11  com.canon.edsdk               	0x000000010630f28a CPtpLinuxUSBDevice::EnumTargetDevice() + 124
12  com.canon.edsdk               	0x000000010630f1c5 CPtpLinuxUSBDevice::GetDeviceCount() + 47
13  com.canon.edsdk               	0x0000000106335906 CLinuxDeviceListManager::EnumChildren() + 84
14  com.canon.edsdk               	0x00000001062fc587 CEdsObject::GetChildCount(unsigned int*) + 101
15  com.canon.edsdk               	0x000000010630497f CMacIOManager::CountPtpCameraPortInfo(unsigned int*) + 107
16  com.canon.edsdk               	0x0000000106302b14 CIOManager::CountCameraPortInfo(unsigned int*) + 28
17  com.canon.edsdk               	0x000000010630b7c2 CEdsCameraList::EnumChildren() + 92
18  com.canon.edsdk               	0x00000001062fc587 CEdsObject::GetChildCount(unsigned int*) + 101
19  com.canon.edsdk               	0x00000001062fd89d CEdsdk::GetChildCount(__EdsObject*, unsigned int*) + 95
20  com.canon.edsdk               	0x00000001062f81e5 EdsGetChildCount + 65
21  info.v002.Camera-Live         	0x00000001061ba9a5 0x1061b8000 + 10661
22  com.canon.edsdk               	0x0000000106303ef4 CMacIOManager::HandleIccEvent(unsigned int, void*) + 72
23  com.apple.ImageCaptureCore    	0x00007fff3ca4c151 -[ICMasterDeviceBrowser addDevice:] + 1364
24  com.apple.ImageCaptureCore    	0x00007fff3ca4b818 -[ICMasterDeviceBrowser updateDevices:type:] + 2559
25  com.apple.ImageCaptureCore    	0x00007fff3ca4abb6 -[ICMasterDeviceBrowser processDevices:] + 253
26  com.apple.ImageCaptureCore    	0x00007fff3ca49a95 __ICPerformBlockOnMainThreadAndWaitUntilDone_block_invoke + 55
27  com.apple.CoreFoundation      	0x00007fff39c06bac __CFRUNLOOP_IS_CALLING_OUT_TO_A_BLOCK__ + 12
28  com.apple.CoreFoundation      	0x00007fff39c06af4 __CFRunLoopDoBlocks + 379
29  com.apple.CoreFoundation      	0x00007fff39c059eb __CFRunLoopRun + 912
30  com.apple.CoreFoundation      	0x00007fff39c04ffe CFRunLoopRunSpecific + 462
[... 429 lines cut ...]
USB Device: USB3.1 Hub
USB Device: Hub
USB Device: USB3.1 Hub
USB Device: Dell Universal Dock D6000
USB Device: USB 3.1 Bus
USB Device: USB2.1 Hub
USB Device: Hub
USB Device: Composite Device
USB Device: USB2.0 Hub
USB Device: D6000 Controller
USB Device: USB2.1 Hub
USB Device: Canon Digital Camera
USB Device: Apple T2 Bus
USB Device: Touch Bar Backlight
USB Device: Touch Bar Display
USB Device: Apple Internal Keyboard / Trackpad
USB Device: Headset
USB Device: Ambient Light Sensor
USB Device: FaceTime HD Camera (Built-in)
USB Device: Apple T2 Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 47.4
Thunderbolt Bus: MacBook Pro, Apple Inc., 47.4
```

Strange part is that the status _does_ change to ```Active```, but then after about 5 seconds the application crashes (as opposed to the behaviour of Camera Live build 13 as described in issue https://github.com/v002/v002-Camera-Live/issues/85 .

bangnoise, 2020-04-10:
Closing this as dup of #86

closed 2020-04-10 by bangnoise