v002/v002-Camera-Live #40: Connection Issue with Ecamm Update

JBrockman1, 2018-08-14:
Looking for a little help?  Ecamm did an update today that has caused Camera Live to stop working.  Prior to this update, things worked perfectly, even up to 30 seconds before.  Not exactly sure what the bug is, but even tried deleting both Ecamm Software and Camera Live and redownloading.  (Of course after restarting everything, checking all batteries, extensions, cords, etc.) Camera Live recognizes that a camera is plugged in, but you can't go any further than that. It freezes with the comment "starting."  Eventually you receive the message that it as unexepectedly stopped working and wants permission to send a report to Apple.  (Additionally the camera goes black and you can't see anything on the preview screen. Unplug the camera from Camera Live and the video function works perfectly.). Any help would be greatly appreciated!  Supposed to be broadcasting a live conference!  Thanks- Jenny

bangnoise, 2018-08-14:
Have you checked the [troubleshooting](https://github.com/v002/v002-Camera-Live/#troubleshooting) section of the README?

When the window pops up asking if you want to send a bug report to Apple - click the button to show details and paste the report here.

JBrockman1, 2018-08-14:
Sorry - it took me a while to get it to give me the text.  The way I've gotten it to work 1 in 15 times is to open Ecamm.  Turn on the camera.  Open Camera Live.  Plug in the camera.  When it freezes, unplug and plug in again.  After about 7 or 8 or times, it will capture.  Sorry, it's worked phenomenally up until today!

JBrockman1, 2018-08-14:
Here's the code it gave me today:  Process:               Camera Live [599]
Path:                  /Users/USER/Desktop/Camera Live-2.app/Contents/MacOS/Camera Live
Identifier:            info.v002.Camera-Live
Version:               8 (8)
Code Type:             X86 (Native)
Parent Process:        ??? [1]
Responsible:           Camera Live [599]
User ID:               501

Date/Time:             2018-08-14 17:03:07.426 -0600
OS Version:            Mac OS X 10.13.6 (17G65)
Report Version:        12
Bridge OS Version:     3.0 (14Y664)
Anonymous UUID:        50278456-372C-882C-4C30-8463CDCBB745

Time Awake Since Boot: 4000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       EXC_I386_GPFLT
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [0]

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   com.canon.edsdk               	0xb0050efd typeinfo name for CMacSharedModule + 13

Thread 1:
0   libsystem_malloc.dylib        	0xa7441da5 small_free_list_remove_ptr_no_clear + 523
1   libsystem_malloc.dylib        	0xa7436ad7 small_free_list_remove_ptr + 163
2   libsystem_malloc.dylib        	0xa7442101 free_small + 771
3   libsystem_malloc.dylib        	0xa743628e szone_free_definite_size + 231
4   libsystem_malloc.dylib        	0xa744408d default_zone_free_definite_size + 57
5   libsystem_malloc.dylib        	0xa7435e66 free + 386
6   com.canon.DppMath             	0x08613aa7 ___kmp_free + 27
7   com.canon.DppMath             	0x0861f31e __kmp_cleanup + 128
8   com.canon.DppMath             	0x0863e19e __kmp_launch_monitor(void*) + 1260
9   libsystem_pthread.dylib       	0xa75914d5 _pthread_body + 347
10  libsystem_pthread.dylib       	0xa759137a _pthread_start + 357
11  libsystem_pthread.dylib       	0xa7590a56 thread_start + 34

Thread 2:: com.apple.NSEventThread
0   libsystem_kernel.dylib        	0xa73d2402 mach_msg_trap + 10
1   libsystem_kernel.dylib        	0xa73d1a3f mach_msg + 47
2   com.apple.CoreFoundation      	0x9377ea68 __CFRunLoopServiceMachPort + 296
3   com.apple.CoreFoundation      	0x9377da56 __CFRunLoopRun + 2262
4   com.apple.CoreFoundation      	0x9377ce51 CFRunLoopRunSpecific + 641
5   com.apple.CoreFoundation      	0x9377cbba CFRunLoopRunInMode + 122
6   com.apple.AppKit              	0x914aa580 _NSEventThread + 165
7   libsystem_pthread.dylib       	0xa75914d5 _pthread_body + 347
8   libsystem_pthread.dylib       	0xa759137a _pthread_start + 357
9   libsystem_pthread.dylib       	0xa7590a56 thread_start + 34

Thread 3:
0   libsystem_kernel.dylib        	0xa73dc716 __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0xa7590e39 _pthread_wqthread + 992
2   libsystem_pthread.dylib       	0xa7590a32 start_wqthread + 34

Thread 4:
0   libsystem_kernel.dylib        	0xa73dc716 __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0xa7590e39 _pthread_wqthread + 992
2   libsystem_pthread.dylib       	0xa7590a32 start_wqthread + 34

Thread 5:
0   libsystem_kernel.dylib        	0xa73dc716 __workq_kernreturn + 10
[... 309 lines cut ...]
VM_ALLOCATE                        176K       22 
__DATA                            14.3M      245 
__FONT_DATA                          4K        2 
__GLSLBUILTINS                    2588K        2 
__IMPORT                             4K        2 
__LINKEDIT                        83.2M       22 
__OBJC                            3276K       95 
__TEXT                           220.0M      248 
__UNICODE                          560K        2 
mapped file                      300.7M      202 
shared memory                      696K       15 
===========                     =======  ======= 
TOTAL                            882.6M      995 

Model: MacBookPro13,2, BootROM MBP132.0247.B00, 2 processors, Intel Core i5, 2.9 GHz, 16 GB, SMC 2.37f20
Graphics: Intel Iris Graphics 550, Intel Iris Graphics 550, Built-In
Memory Module: BANK 0/DIMM0, 8 GB, LPDDR3, 2133 MHz, 0x802C, 0x4D5435324C31473332443450472D30393320
Memory Module: BANK 1/DIMM0, 8 GB, LPDDR3, 2133 MHz, 0x802C, 0x4D5435324C31473332443450472D30393320
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x157), Broadcom BCM43xx 1.0 (7.77.37.31.1a9)
Bluetooth: Version 6.0.7f10, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB 3.0 Bus
USB Device: iBridge
Thunderbolt Bus: MacBook Pro, Apple Inc., 33.1
Thunderbolt Bus: MacBook Pro, Apple Inc., 33.1

JBrockman1, 2018-08-14:
Here are a few screenshots:  
[image]
[image]
[image]

bangnoise, 2019-11-19:
Please try with [the latest release](https://github.com/v002/v002-Camera-Live/releases/latest).

closed 2019-12-02 by bangnoise