v002/v002-Camera-Live #41: Camera is "Active" but MacOS cannot access it

warent, 2018-08-27:
My camera appears as "Active" in the app, the camera is powered on and everything, but it's not an option in any apps (e.g. Quicktime, Hangouts, etc.). Is there something I'm missing?

Thanks!

[image]
[image]

vade, 2018-08-28:
That those applications do not support Syphon :) Syphon applications are an eco-system of opt-in software tools. Google Hangouts and Quicktime have not been extended to support Syphon.

bangnoise, 2018-08-28:
[CamTwist](http://camtwiststudio.com) might help - I haven't tried it myself but people seem to use it.

mallton, 2018-11-05:
bangoise is right, you need CamTwist. Once installed, click on View -> Main Window -> Click on Syphon and then click on Select. In the Settings column select CamTwist in the drop down list of Syphon Server options.

Then, in whatever app you open, you will simply change the camera source to CamTwist.

More info and screenshots here: [https://www.lightscameralive.com/blog/how-to-use-a-dslr-camera-to-make-your-live-video-magical](url)

closed 2019-11-19 by bangnoise