v002/v002-Camera-Live #42: Canon T2i/550D doesn't connect

sandeld, 2018-09-02:
This is the first time I'm trying to use your program so I'm not sure if it would work with a previous version (as I've seen you mention in other support threads).

I've updated to the latest firmware on my camera (per your troubleshooting guide) and am trying to run Camera Live 2 (version 9), but it never recognizes my camera. 

I'm using a Macbook Pro running High Sierra 10.13.6

simonhaenisch, 2019-08-24:
Same camera, got it working in the following steps in that exact order:

- Camera is off
- Connect camera to USB port
- Turn camera into movie mode
- Turn camera on
- Start Camera Live - nothing will show up
- Quit Camera Live
- Start Camera Live again - now you'll hear the camera click and it'll show up in the list
- Select it and wait for it to become active (then it can be accessed with a syphon client like OBS)
- If you click somewhere else in the list, you will deselect it again which will deactivate it also

bangnoise, 2019-11-19:
Should work with [the latest release](https://github.com/v002/v002-Camera-Live/releases/latest).

closed 2019-11-19 by bangnoise