v002/v002-Camera-Live #44: Canon t3i is 4:3 aspect ratio

servantworker, 2018-10-27:
Letterboxes when trying to set a resolution other than 16:9 (seen in both Camtwist and Syphon Recorder)

bangnoise, 2018-11-05:
Do you see a full 16:9 image with black borders to make it 4:3 or a cropped 16:9 image (no black borders, edges missing)?

Camera Live just emits what the camera sends it, which generally corresponds to the image shown in Live View on the camera.

delphiy, 2019-07-18:
I am getting black bars left and right side quicktime - Mac

bangnoise, 2019-07-19:
What do you mean by QuickTime? What's your workflow? Camera Live > ?

bangnoise, 2019-07-19:
CamTwist outputs at a fixed resolution (see CamTwist > Preferences... > General > Video Size). To eliminate the bars, set that resolution to match the output from the camera. The Syphon [Simple Client](https://github.com/Syphon/Simple/releases/tag/5) will display the resolution for a server - to eliminate the bars from CamTwist, match the size Simple Client reports.

closed 2019-07-19 by bangnoise

delphiy, 2019-07-19:
I have set 1920 x 1080 in camtwist preferences.
 My camera is canon dslr 4000d , what resolution should I set in camtwist. 
Black bars still there

bangnoise, 2019-07-19:
From the README (please read it before posting)

> The camera sends the image used for Live View - the dimensions will not match the camera's movie recording settings.

From literally the post you replied to:

> The Syphon Simple Client will display the resolution for a server - to eliminate the bars from CamTwist, match the size Simple Client reports.

delphiy, 2019-07-19:
Make it simple please ,
My camera canon d4000 . I am not sure how I can get image resolutions 

What resolution values should I put in camtwist

bangnoise, 2019-07-19:
1. Connect camera, start Camera Live
2. Download [Simple Client](https://github.com/Syphon/Simple/releases/download/5/Syphon.Simple.Apps.zip)
3. Open Simple Client, your camera should be displayed - if not, select it from the menu in the toolbar
4. Read the dimensions from the status display in Simple Client's toolbar
5. Set those values in CamTwist > Preferences... > General > Video Size > Custom

This is a CamTwist usage issue, please contact them for further support.