v002/v002-Camera-Live #45: Connection to 77d stuck on "Starting"

ACRZ, 2018-11-02:
I don't know what happened but everything was working perfectly fine earlier today.  The only thing that has changed is that I have purchased the full version of ECAMM.  Now whenever I try to connect to my camera it just says starting.  
[image]

Doormandean, 2018-11-04:
Yep, I have exactly the same issue, it seemed stronger when upgrading to Mojave.

I really hope someone can fix it, but did I see that the app isn't getting any further development?

Doormandean, 2018-11-04:
This was such a fantastic app, and while there are options with things like Elgato's Cam Link, they don't support autofocus, which Camera Live did incredibly well. I now have to completely re-think how I'm going to live stream, or even if I need to buy completely different cameras, unless the developers are aware of it and are actually working on a fix?

Some feedback would be massively appreciated!

mallton, 2018-11-04:
From my research, the issue has to do with the new System Integrity Protection in MacOS Mojave. Unfortunately, the only way currently to use Camera Live seems to be to disable SIP entirely. :( Here's how:

Click the 🍎 symbol in the Menu bar.
Click Restart...
Hold down Command-R to reboot into Recovery Mode.
Click Utilities
Select Terminal
Type csrutil disable
Press Return or Enter on your keyboard
Click the 🍎 symbol in the Menu bar
Click Restart

Doormandean, 2018-11-04:
> From my research, the issue has to do with the new System Integrity Protection in MacOS Mojave. Unfortunately, the only way currently to use Camera Live seems to be to disable SIP entirely. :( [...]

Thank you for the input, seems a terrifying way of making it work though? Presumably it leaves your system open to all kinds of potential issues?

e13o13, 2018-11-12:
I am having this same issue after switching to Mojave - is disabling the SIP the accepted best solution at this point?

bangnoise, 2018-11-16:
I'm not sure SIP is involved in this at all - that it worked after disabling it may be a coincidence.

Unfound45, 2018-12-03:
Any progress on fixing this? Would love to use this app! :)

Unfound45, 2018-12-03:
Just found a work around!

Plug DSLR in (turned off)
Open Camera Live
Turn DSLR on
*If camera live automatically selects your camera (highlighting it blue) turn DSLR off and wait for DSLR to become unlisted*
Turn DSLR Back on *wait*
DSLR should be "unselected" 
Double click DSLR (not single click)

This has worked for me now 10 times in a row. The app has crashed while doing this processes a few times but once you can get it down it seems to work!

dank81, 2018-12-07:
This seems to happen when there are images stored on the camera's memory card. Try deleting the images, removing the card or changing to an empty/blank card.

Another one - if the camera's LCD display says "busy" when you try to activate Camera Live, try disconnecting and re-connecting the USB cable.

ToneBekkestad, 2019-03-01:
> Just found a work around!
>
> Plug DSLR in (turned off)
> Open Camera Live
> Turn DSLR on
> _If camera live automatically selects your camera (highlighting it blue) turn DSLR off and wait for DSLR to become [...]

What kind of camera do you use? 
I was so happy to see your work-around, and have just tried it, but it does not work for me.
I have a Canon EOS 80D.
When I turn the DSLR off and then on again, my camera is automatically selected (it is not "unselected"), which is maybe the reason why double-clicking it does not work.
Any ideas?

ToneBekkestad, 2019-03-01:
I have the same problem. 
I was so happy when I heard of Camera Live and the possibilities to go live with my DSLR, but it does not work for me - it stops at "Starting" and never continues.
I have a Canon EOS 80D and a Mac with MacOS Mojave.

Any suggestions to how I can get it to work?
(I have tried removing the SIP, the cameras WiFi-connection, and any images/movies on the camera).

Thanks!

AbeSaldana, 2019-03-21:
I change the SIP to disable, but still, have the same issue getting stuck on the "Starting", so I don't think this is the main solution on how to make it work. I'm using a Canon EOS M50, it was working yesterday and now is not connecting at all, no updates or anything different today, also I have a brand new card and no images on it.

bangnoise, 2019-03-21:
I've deleted comments suggesting that SIP be disabled. SIP is not related to this issue. Do not disable SIP.

shaunnorris, 2019-05-10:
In what may be useful information in solving this, when i run Camera Live with my 12" Macbook (2018 vintage) it works fine.  Using same USB-C dongle, same USB cable, and same Canon 760D camera on my 13" Macbook Pro (also 2018 vintage) it doesn't work and either shows "Camera Error" or "Starting" indefinitely. Possibly suggests some USB hardware incompatibility with certain versions of Macbook / Macbook Pro? Configs from system report on both below.  Hopefully this helps the devs on this project look for a fix? Thx, Shaun
[image]
[image]

ksigWyatt, 2019-06-30:
I am also having this issue. And have tried the mentioned workaround from @Unfound45, with no luck. I have an EOS Mk 3. Using `v9` of Camera Live on MacOS `10.14.5` and EOS Utility `v3.10.20 (3.10.20.0)`.

Camera Live is stuck saying Starting whenever the camera is connected.

This could be related to what @bangnoise was saying about the Utilities.
> Weird - there are reliability issues with the underlying Canon SDK on recent macOS versions, and perhaps you happen to have hit an issue now.

KristjanToopEG, 2019-08-12:
Same problem here. Anyone found a solution for it?

Markaplan, 2019-08-23:
Solved this issue on my mac pro 2013 with Mojave 10.14:

I just ran the app with root premissions. 
Open terminal and type: 
sudo "/Applications/Camera Live.app/Contents/MacOS/Camera Live" 
Then it`ll ask your password and then should run. 

Worked out for me, hope it`ll help. 

The path to the app could be different depending on where you put it :D

lowfideluxe, 2019-10-05:
> Solved this issue on my mac pro 2013 with Mojave 10.14:
> 
> I just ran the app with root premissions.
> Open terminal and type:
> sudo "/Applications/Camera Live.app/Contents/MacOS/Camera Live"
> Then it`ll ask your password and then should run.
> 
> Worked out for me, hope it`ll help.
> 
> The path to the app could be different depending on where you put it :D
> 
> P.S.: it didn`t work on the second time when I ran it, so I just restarted the app through terminal several times and it finally worked again :D

THAT worked FINE! THNX

montikid, 2019-10-10:
I can't seem to work this through terminal. I keep getting a "No such file or directory" message." This is so frustrating. I've only been able to use this once. Has anyone found an alternative? I want to activate QuickTime movie recording using my dslr camera as video input and my external mic as audio input. I've only been able to do it once and ever since, I've been having the same Camera Live starting error.

bangnoise, 2019-11-19:
Please try with [the latest release](https://github.com/v002/v002-Camera-Live/releases/latest).

closed 2019-12-02 by bangnoise

hwschuur, 2020-03-31:
Don't know how to get Camera Live to work reliably. It worked for some time, but now it doesn't anymore. Keep getting 'Camera error' after 'Starting', or Camera Live hangs at 'Starting'.

I'm on Catalina 10.15.4. When I run the app via `sudo` in the Terminal, I get
`2020-03-31 20:26:25.234 Camera Live[1094:20941] get NSURLIsDirectoryKey Error : Error Domain=NSCocoaErrorDomain Code=260 "The file “ic_hevcdec.framework” couldn’t be opened because there is no such file." UserInfo={NSURL=Contents/Frameworks/ic_hevcdec.framework -- file:///Applications/Camera%20Live.app/, NSFilePath=/Applications/Camera Live.app/Contents/Frameworks/ic_hevcdec.framework, NSUnderlyingError=0x7fd59f897430 {Error Domain=NSPOSIXErrorDomain Code=2 "No such file or directory"}}
sudo "/Applications/Camera Live.app/Contents/MacOS/Camera Live"'

lowfideluxe, 2020-03-31:
Hi @hwschuur , 
Im working with Camera Live with a similar setup for zoom use with camera twist. I also had the same problems. this is how i make it work:
1: attach the camera via usb
2: power on the camera
3: open up Camera Live - (mine get stuck at starting)
4: open terminal and run the app via sudo (a second Camera Live opens)
5: in the secnd Camera Live windows- try to operate the camera.
It worked for me

hwschuur, 2020-03-31:
Hi @lowfideluxe, thanks for your suggestion. I tried it and everything is well until step 5: there's no camera listed in the second Camera Live window.

Could this be related to the Catalina 10.15.4 update @bangnoise?

bangnoise, 2020-03-31:
@hwschuur please can you try [the current alpha release](https://github.com/v002/v002-Camera-Live/releases/download/13/Camera.Live.zip). If the problem persists please open a new issue.

hwschuur, 2020-04-01:
@bangnoise Thanks. I did some testing, please see https://github.com/v002/v002-Camera-Live/issues/85

Keturah117, 2020-04-20:
So I have finally gotten past the "starting" issue, but now my camera Apple Mac won't recognize the camera as the webcam. Every time I open an application it says that there is no camera found or connected. Need help trying to do live streams for my church! Thank you for your time.

lowfideluxe, 2020-04-21:
@Keturah117 
What software are you using to do live streams? 
If you are using OBS, hangouts, zoom , you could use camtwist to conect cameralive as a Syphon client. I mean: cameralive let you conect your dslr, but in order to get the camera feed from the DSLR to your streaming software, cameratwist can be used as a bridge that comunicates the two softwares

Keturah117, 2020-04-21:
I use streamyard but I haven't gotten that far because I can't even get it to connect in photo booth. My apologies I should have mentioned I do use camtwist and it lets me select camera live but that's it.  When i try to use photo booth it won't work so i didn't even try to use streamyard.

Dr3wskjie, 2020-05-08:
> Just found a work around!
>
> Plug DSLR in (turned off)
> Open Camera Live
> Turn DSLR on
> _If camera live automatically selects your camera (highlighting it blue) turn DSLR off and wait for DSLR to become [...]

This worked for me, but running the App would have to go through "Terminal". I am so relief that this has helped me out a lot. "HOWEVERRRR" :) my Canon 7D is not able to do any auto-focusing. Everytime I connect my Canon Utility, it conflicts with Camera Live and it crahes/freezes. Any inputs on this ? or Anyone ??

dineshappavoo, 2020-05-08:
Keeping the camera on/connected, and restarting the computer helped in some cases. Otherwise restarting the app many times works.

Dr3wskjie, 2020-05-08:
> Keeping the camera on/connected, and restarting the computer helped in some cases. Otherwise restarting the app many times works.

I haven't disconnected the camera at all. Restarting the App sometimes works, and sometimes it doesn't. I really wonder why the creator of this App is not working in finding a solution or some type... 

Thank you so much for your input :D

bangnoise, 2020-05-08:
There are so many different problems described in this issue - some of them more human than technical - **read** the [Troubleshooting](https://github.com/v002/v002-Camera-Live/#troubleshooting) section of the README and **try everything it describes**.

bangnoise, 2020-05-08:
also some of you need to read the [Installation & Use](https://github.com/v002/v002-Camera-Live/#installation-and-use) section too - Camera Live is not a webcam driver.

CFPro, 2020-05-25:
Updating my camera firmware solved all my problems

Dr3wskjie, 2020-05-26:
I did the same as well and it's been working flawlessly. Thank you so much
for the update.

*Drew*

dineshappavoo, 2020-05-26:
`+1`. Updating the camera firmware fixed the `Starting` issue—I used canon 80D. Camera live works like a charm now. 

One hiccup is that I had to restart CameraLive when I lock the computer and login again. I disabled the camera sleep mode, but still the problem persists—perhaps some sleep mode in CameraLive or somewhere else. Again this is not related to the `Starting` issue—this original ticket is for.

Dr3wskjie, 2020-06-01:
Okay, 
I never thought I'd run into but here it is... 
Camera Live - Still detected my camera (showing _Active_) 
Cam Twist - Still showing Cam Live

However..... I have been using ZOOM for all my meetings and today, with all the invitations I had received for a meeting, it kept asking me to **UPDATE** the Application, so I had to do it. (fingers crossed). 

Guess what? When I opened up ZOOM, with Cam Live being active and everything, ZOOM is **not showing** the Cam Live as an option... I feel like I back-tracked 50% from my original issue (not able to get Cam Live Active). It was so working beautifully until the update for ZOOM.

Has anyone ran into this issue recently ? 

PLEASE help if so. 

Thank you so much in advance. 

*Drew*

lowfideluxe, 2020-06-01:
> Okay,
> I never thought I'd run into but here it is...
> Camera Live - Still detected my camera (showing _Active_)
> Cam Twist - Still showing Cam Live
>
> However..... I have been using ZOOM for all my [...]

Hi! Im also using camera live with zoom, i have to update to keep using it for my online classes and i also lost the option to use it with cam twist. 
I found this and its working now. (im using Zoom 5.0.4)

https://www.theandroidsoul.com/how-to-fix-virtual-camera-not-working-problem-on-zoom/#method-1-remove-official-signature-from-zoom-app-on-mac

Dr3wskjie, 2020-06-02:
> Hi! Im also using camera live with zoom, i have to update to keep using it for my online classes and i also lost the option to use it with cam twist.
> I found this and its working now. (im using Zoom [...]

Thank You so very much for the tip. I have tried it and it's working again. Hopefully it continues to stay this way unless further updates (that may cause the same problem). 

Much appreciate the quick tip. 

Have a great and safe night.

Dr3wskjie, 2020-06-20:
> Thank You so very much for the tip. I have tried it and it's working again. Hopefully it continues to stay this way unless further updates (that may cause the same problem).
>
> Much appreciate the [...]

> Hi! Im also using camera live with zoom, i have to update to keep using it for my online classes and i also lost the option to use it with cam twist.
> I found this and its working now. (im using Zoom [...]

_By any chance, do you know if your method/solution will continue to work if ZOOM is being updated continuously ? (i.e. per your advise, it's for ver. 5.0.4 and the latest update is 5.1.0) Any advice/comments/recommendations would be greatly appreciated._