v002/v002-Camera-Live #47: Stuck on starting

stXwinter, 2018-11-07:
Hello? Is anyone going to reply to my previous Closed Thread question?

Cyberpunk110, 2018-11-21:
Hi, try switching to live View Mode on your camera.  
Little tutorial to show what i mean: 
https://support-sg.canon-asia.com/contents/SG/EN/8202785200.html

worked for me! Good luck

dank81, 2018-12-07:
This seems to happen when there are images stored on the camera's memory card. Try deleting the images, removing the card or changing to an empty/blank card.

Another one - if the camera's LCD display says "busy" when you try to activate Camera Live, try disconnecting and re-connecting the USB cable.

Yashkin0, 2018-12-30:
I am having an issue with this as well, my memory card is empty as suggested, but it just says starting. On the LCD screen there is a picture of a laptop. I am using a t6i and a macbook pro with the latest release of Mojave

CookieDude2332, 2019-01-22:
I too am having this same issue with my Canon 700D. Sometimes the screen will tell me to press the live view button to get it working but all my buttons stop working.

rdmorton, 2019-01-22:
It's working well on my Canon 70D.  Have you updated the firmware on your Canon?

rdmorton, 2019-01-22:
I'm using High Sierra, haven't tested this with Mojave.

ksigWyatt, 2019-06-30:
I'm having this issue also. See my comment on issue #45.

bangnoise, 2019-11-19:
Please try with [the latest release](https://github.com/v002/v002-Camera-Live/releases/latest). Closing as this issue is a duplicate of #45.

closed 2019-11-19 by bangnoise