v002/v002-Camera-Live #48: Canon EOS R not working on Mac

WonderWomansGirlfriend, 2018-11-26:
Hey!! Ive been trying for the longest time to get my Canon EOS R connected on 2 different systems but it is not going active.  Is the new EOS R supported is that why it's not working?   If not when will you be adding support?  Thanking you in advance. Cheers Faith

xjohnson2018, 2018-11-27:
Good morning, 
 This seems to be a problem widespread problem possibly with the update of MacOS. Just a heads up.

WonderWomansGirlfriend, 2018-11-28:
oh ok cool thanks for letting me know!! 👍

dank81, 2018-12-07:
If you're stuck with Camera Live saying "starting", try deleting any stored images from the camera's memory card. That worked for me with my 5D Mk IV.

Snooz82, 2018-12-21:
I have also an EOS R and it does not work.
macOS Mojave

xjohnson2018, 2018-12-21:
The following has worked for me:
Plug DSLR in (turned off)
Open Camera Live
Turn DSLR on
If camera live automatically selects your camera (highlighting it blue) turn DSLR off and wait for DSLR to become unlisted
Turn DSLR Back on wait
DSLR should be "unselected"
Double click DSLR (not single click)

DarrinJasz, 2019-01-02:
I had a similar issue of it working only about 10% on the time on Mojave.  I ultimately had to buy another Mac and down-grade it to High Sierra for it to work. Hopefully someone is working on a fix :-)

bangnoise, 2019-11-19:
The EOS R is supported in [the latest release](https://github.com/v002/v002-Camera-Live/releases/latest).

closed 2019-11-19 by bangnoise

Snooz82, 2019-11-19:
Awesome! 😎