v002/v002-Camera-Live #49: Autofocus not working

apedronio, 2018-11-29:
Rebel T6 set on video mode, I have to mid-press the shutter on the camera for it to refocus. 

Cheers! And great app btw!

Burotica, 2020-03-30:
69/5000
I have the same problem on my Nikon D5300, the autofocus does not work.

bangnoise, 2020-03-30:
Yeah there is no support for autofocus at the moment - note that most still cameras don't have continuous autofocus so even if/when it is implemented it will take time to focus.

Depending on the camera you may be able to use the shutter button to active auto-focus as usual.

ykbryan, 2020-03-31:
I tried on XT-3.
It seems that it works without autofocus. I tried to see if it works half-pressing the shutter button but it doesn' t.