v002/v002-Camera-Live #50: Canon M50 thin micro HDMI to MacBook Pro

jkn1114, 2019-02-02:
I have my thin micro HDMI cable connected to from my Canon M50 to my MacBook Pro mid 2018 15 inch. The camera live doesn't even detect it. I purchased this cable because I want to go from the thin micro USB to HDMI for higher quality. I have no idea why the camera live is not detecting the camera. I did all of the normal steps as I do each time. ANY help would be greatly appreciated. Or I'll have to simply return the cable to Best Buy and be stuck with less video quality.

dank81, 2019-02-02:
Camera Live runs over USB only. It doesn’t run over HDMI.

If you want to use HDMI, you need an HDMI capture device like the BlackMagic Designs Mini Recorder.

dank81, 2019-02-02:
... and if you use that, you don’t need Canon Camera Live nor Camera Live Syphon (i.e. this project)

commit 69478a95 (2019-02-04): Update README.md re USB not HDMI
Fix #50
closed 2019-02-04 by bangnoise