v002/v002-Camera-Live #51: Nikon 3200 is not detected on Mac

sifus, 2019-04-14:
Hi,

my DSLR Nikon d3200 is connected to my mac with USB but when i launch Camera Live, ma camera is find...

bangnoise, 2019-04-14:
See README.md - Canon cameras only

mydudechris, 2020-03-27:
Will you offer support for Nikon at any point in the future?

bangnoise, 2020-03-27:
Which version are you using? The [alpha version 13](https://github.com/v002/v002-Camera-Live/releases/download/13/Camera.Live.zip) has support for more cameras.

mydudechris, 2020-03-27:
I'm using Alpha V.13. It recognizes my Nikon D750, connected over USB for a moment then the image freezes in place :(

mydudechris, 2020-03-27:
Not sure if this is helpful: https://sdk.nikonimaging.com/apply/

bangnoise, 2020-03-27:
I'm not about to add support for Nikon's own SDK (but of course pull requests are welcome). It's encouraging that you get a frame from the D750 - I'm aiming to improve support for non-Canon cameras once I have the rebuild using libgphoto2 stabilised.

mydudechris, 2020-03-27:
I'm kind of new coming into this, alternatively, I could use something like the Blackmagic web presenter, right? I'd much rather use something like this!

bangnoise, 2020-03-27:
@mydudechris It would be helpful if you could

1. open Console (in /Applications/Utilities) and search for "Camera Live" (type "Camera Live" into the search box followed by a return
2. Launch Camera Live with your Nikon connected and try to start it and wait for it to emit its single frame
3. Copy and paste the Console messages you get here

bangnoise, 2020-03-27:
I can't speak for the connectivity of Nikon cameras but it sounds like something like the Web Presenter should work if your camera has HDMI output - one thing to be aware of is that some cameras overlay their viewfinder UI over the HDMI output - I have no idea if that affects your camera or not.

mydudechris, 2020-03-27:
Hmm, interesting, now it appears to work flawlessly. The one big change that I made was connecting my camera to a continuous power source, whereas before I had a fully charged battery. Very strange. Anyway, the app appears to be working although I see 2 instances of the camera in the camera live list (Nikon DSC D750). 

Here's some output from console: 
`
default	14:49:21.609865-0400	hidd	Connection added: IOHIDEventSystemConnection uuid:A133D473-BAE2-4854-AEA1-B8910BFE31FF pid:883 process:Camera Live type:Passive entitlements:0x0 caller:HIToolbox: ___GetIOHIDEventSystemClient_block_invoke + 26 attributes:(null) state:0x0 events:0 mask:0x0
default	14:49:21.861076-0400	hidd	Connection added: IOHIDEventSystemConnection uuid:E5FA0ACD-08E3-48EC-965D-85E37DBCAFC2 pid:883 process:Camera Live type:Passive entitlements:0x0 caller:HIServices: ___GetIOHIDEventSystemClient_block_invoke + 26 attributes:(null) state:0x0 events:0 mask:0x0
default	14:49:22.296173-0400	kernel	000245.795667 Camera Live@(null): AppleUSBHostUserClient::start: missing entitlement com.apple.ibridge.control
default	14:49:22.296431-0400	kernel	000245.795927 Camera Live@(null): AppleUSBHostUserClient::start: missing entitlement com.apple.ibridge.control
default	14:49:22.296673-0400	kernel	000245.796168 Camera Live@(null): AppleUSBHostUserClient::start: missing entitlement com.apple.ibridge.control
default	14:49:22.296912-0400	kernel	000245.796409 Camera Live@(null): AppleUSBHostUserClient::start: missing entitlement com.apple.ibridge.control
default	14:49:22.297158-0400	kernel	000245.796655 Camera Live@(null): AppleUSBHostUserClient::start: missing entitlement com.apple.ibridge.control
default	14:49:22.297538-0400	kernel	000245.797034 Camera Live@(null): AppleUSBHostUserClient::start: missing entitlement com.apple.ibridge.control
default	14:49:22.992240-0400	kernel	AGC:: [Camera Live pid:883 mux-aware] starting, non-mux-aware app count 0 WS-ready 1 switch-state 0 IG FBs 0 EG FBs 3:1f power-state 0 3D idle HDA busy system-state 1 power-level 20:20 power-retry 0:0 connect-change 0
default	14:49:22.992314-0400	kernel	AGC:: [Camera Live pid:883 non-mux-aware] starting, non-mux-aware app count 1 WS-ready 1 switch-state 0 IG FBs 0 EG FBs 3:1f power-state 0 3D idle HDA busy system-state 1 power-level 20:20 power-retry 0:0 connect-change 0
default	14:49:23.002422-0400	Camera Live	Received configuration update from daemon (initial)
default	14:49:23.004152-0400	kernel	000246.503625 Camera Live@(null): AppleUSBHostUserClient::start: missing entitlement com.apple.ibridge.control
default	14:49:23.004381-0400	kernel	000246.503856 Camera Live@(null): AppleUSBHostUserClient::start: missing entitlement com.apple.ibridge.control
default	14:49:23.004606-0400	kernel	000246.504081 Camera Live@(null): AppleUSBHostUserClient::start: missing entitlement com.apple.ibridge.control
default	14:49:23.004829-0400	kernel	000246.504305 Camera Live@(null): AppleUSBHostUserClient::start: missing entitlement com.apple.ibridge.control
default	14:49:23.005038-0400	kernel	000246.504513 Camera Live@(null): AppleUSBHostUserClient::start: missing entitlement com.apple.ibridge.control
default	14:49:23.005382-0400	kernel	000246.504857 Camera Live@(null): AppleUSBHostUserClient::start: missing entitlement com.apple.ibridge.control
error	14:49:24.102396-0400	tccd	{ID: info.v002.Camera-Live, PID[883], auid: 501, euid: 501, binary path: '/private/var/folders/tx/bggh2yr163l_q6pp6bzsxckr0000gn/T/AppTranslocation/7937D6AE-5616-4AE3-A380-04F447DA1B2D/d/Camera Live.app/Contents/MacOS/Camera Live'} attempted to call TCCAccessRequest without the com.apple.private.tcc.manager.check-by-audit-token entitlement
default	14:49:26.189708-0400	runningboardd	Invalidating assertion 321-143-163 (target:executable<Camera Live(501)>) from originator 143
default	14:49:26.217832-0400	runningboardd	[executable<Camera Live(501)>:883] Ignoring jetsam update because this process is not memory-managed
default	14:49:26.218003-0400	runningboardd	[executable<Camera Live(501)>:883] Set darwin role to: UserInteractiveNonFocal
default	14:49:26.218297-0400	runningboardd	[executable<Camera Live(501)>:883] Ignoring GPU update because this process is not GPU managed
default	14:49:26.312587-0400	tccd	-[TCCDAccessIdentity staticCode]: static code for: identifier info.v002.Camera-Live, type: 0: 0x7fbbbfd23b60 at /var/folders/tx/bggh2yr163l_q6pp6bzsxckr0000gn/T/AppTranslocation/7937D6AE-5616-4AE3-A380-04F447DA1B2D/d/Camera Live.app
default	14:49:31.559477-0400	Camera Live	LSExceptions shared instance invalidated for timeout.
default	14:49:31.844463-0400	loginwindow	-[PersistentAppsSupport saveLogoutPersistentState:finalSnapshot:] |      checkAgainstApp is : Camera Live
default	14:49:31.844482-0400	loginwindow	-[PersistentAppsSupport appShouldBeRelaunched:] | entered. checking app: Camera Live
default	14:49:31.844516-0400	loginwindow	-[PersistentAppsSupport saveLogoutPersistentState:finalSnapshot:] |           Adding to relaunchArray: Camera Live
default	14:49:31.847651-0400	loginwindow	-[PersistentAppsSupport saveLogoutPersistentState:finalSnapshot:] |           Contents:(
        {
        BackgroundState = 2;
        BundleID = "com.obsproject.obs-studio";
        Hide = 0;
        Path = "/Applications/OBS.app";
    },
        {
        BackgroundState = 2;
        BundleID = "com.apple.console";
        Hide = 0;
        Path = "/System/Applications/Utilities/Console.app";
    },
        {
        BackgroundState = 2;
[... cut ...]
default	14:51:18.394898-0400	runningboardd	Invalidating assertion 321-143-174 (target:executable<Camera Live(501)>) from originator 143
default	14:51:18.504446-0400	runningboardd	[executable<Camera Live(501)>:883] Ignoring jetsam update because this process is not memory-managed
default	14:51:18.504498-0400	runningboardd	[executable<Camera Live(501)>:883] Set darwin role to: UserInteractiveNonFocal
default	14:51:18.504703-0400	runningboardd	[executable<Camera Live(501)>:883] Ignoring GPU update because this process is not GPU managed
`

mydudechris, 2020-03-27:
Here are other errors that appeared in the list:
`error	14:49:21.478293-0400	tccd	Prompting policy for hardened runtime; service: kTCCServiceAppleEvents requires entitlement com.apple.security.automation.apple-events but it is missing for ACC:{ID: info.v002.Camera-Live, PID[883], auid: 501, euid: 501, binary path: '/private/var/folders/tx/bggh2yr163l_q6pp6bzsxckr0000gn/T/AppTranslocation/7937D6AE-5616-4AE3-A380-04F447DA1B2D/d/Camera Live.app/Contents/MacOS/Camera Live'}, REQ:{ID: com.apple.appleeventsd, PID[356], auid: 55, euid: 55, binary path: '/System/Library/CoreServices/appleeventsd'}
`

`error	14:46:43.233048-0400	tccd	{ID: info.v002.Camera-Live, PID[766], auid: 501, euid: 501, binary path: '/private/var/folders/tx/bggh2yr163l_q6pp6bzsxckr0000gn/T/AppTranslocation/D6574F3D-8513-4B05-9D13-553387FBA5BC/d/Camera Live.app/Contents/MacOS/Camera Live'} attempted to call TCCAccessRequest without the com.apple.private.tcc.manager.check-by-audit-token entitlement
`

bangnoise, 2020-03-28:
Glad it's working. Nothing remarkable in those logs - was that from a "working" run or one where the image freezes?

mydudechris, 2020-04-01:
@bangnoise please see the attached PDF. It's an export of all the activity from the app, startup to crash where the frame freezes. 

[Untitled spreadsheet - Sheet1.pdf](https://github.com/v002/v002-Camera-Live/files/4416157/Untitled.spreadsheet.-.Sheet1.pdf)

hajdebliem, 2020-04-05:
Hey there, i just played around with the app to see if my D800 would show up via Syphon. It sure did! There is no camera settings control when live . So i had to unplug it, change the setting and the reconnect and relaunch the App. Sadly the Stream gets interrupted quite often…

I might have to check out a more commercial route for my distance learning setup, but i like the possibility to hook up a dslr to mad mapper! 

[syslog_CameraLive.txt](https://github.com/v002/v002-Camera-Live/files/4434712/syslog_CameraLive.txt)

bangnoise, 2020-04-05:
> Sadly the Stream gets interrupted quite often…

@hajdebliem do you mean it stutters/pauses, or stops and doesn't recover?

hajdebliem, 2020-04-05:
It stops and does not recover. Relaunching the app is the only way so far i discovered to get the stream back. 
One wired thing that i noticed is that when i wave my hand in front of the camera or move the camera it tends to freeze immediately. As if it does not like when too many pixels change… no obvious logic here.

bangnoise, 2020-04-05:
@hajdebliem disconnect all other USB devices, do not use a hub but connect directly to the computer, use the shortest USB cable you have and make sure it isn't running beside power cables or other sources of interference. If you have an external power supply for your camera, use that.

trich19, 2020-04-11:
I got my new Nikon d780 to work, it has usb c, if that makes any difference.  About every 5 mins the shutter clicks but it always recovers quite quickly.  Only issue is that the resolution is fairly low, and even when camera is in full manual, it seems to do its own exposure

mbesemann, 2020-04-21:
Working with a Nikon D5200 here!  You're right about the consistent power source.  While running on battery power, I had it working for a couple of minutes and then Camera Live was no longer detecting it (even though the camera was still powered on).  Swapped to a fresh battery and it started working again.  I'll have to invest in a power adapter for my camera if I want to have any longer sessions.

mlim18, 2020-05-01:
Any updates on this working with a D750 and what needs to be done to get it to work? I'm brand new to this.

joshspicer, 2020-05-02:
Got my Nikon D3400 working.  Note I needed to remove the signature (cc #102) in my testing for Zoom to pick up the camera.

jfrux, 2020-05-05:
@mbesemann Still having success with the D5200?  Just plugged mine in today and it says "Active".  I performed `codesign --remove-signature /Applications/zoom.us.app/` and restarted the app but still not showing up in Zoom.

I'm also getting this when the camera connects if it helps...

`default	07:39:13.843030-0400	kernel	1248916.749764 Camera Live@(null): AppleUSBHostUserClient::start: missing entitlement com.apple.bluetooth.control`

ironc0in, 2020-05-08:
I had the same issue, swapped the battery with a one that was full, and it immediately popped up! 🤦🏽‍♂️