v002/v002-Camera-Live #56: Not working on MacOS Mojave

yannickErp, 2019-07-10:
I'm on MacOs 10.14.5 and Camera-Live doesn't recognise my Canon EOS 650D, although it is in the list with the apparently working cameras, and EOS Utility DOES recognise the camera.

If anyone has a fix or an alternative, please answer me.

I see that this git is more or less dead so hopefully someone else will create an alternative soon.

bangnoise, 2019-07-11:
Have you tried the steps in the Troubleshooting section of the README?

> I see that this git is more or less dead so hopefully someone else will create an alternative soon.

me too

dank81, 2019-07-12:
> > I see that this git is more or less dead so hopefully someone else will create an alternative soon.
> 
> me too

Many thanks Tom for this project, I appreciated using it!

For anyone looking for a more robust alternative, I recently switched to using the BlackMagic UltraStudio Mini Recorder, which takes an HDMI input from your camera and connects to your Mac using a Thunderbolt connection. For recent Macs without a Thunderbolt port, you can get a USB 3.0 adapter from Apple. The HDMI output from your camera comes into your Mac as a video capture device via the BlackMagic drivers, and is selectable as a source in OBS and other software.

The full kit I use as follows:

* BlackMagic UltraStudio Mini Recorder (~ $200):
https://www.blackmagicdesign.com/au/products/ultrastudio/techspecs/W-DLUS-04 (~$200)

* Mini HDMI to HDMI cable for the camera (~ $65):
https://shop.usa.canon.com/shop/en/catalog/mini-hdmi-cable-htc-100
(You can get a cheaper one that isn't Canon but I like this one because it fits into the "cable keeper" frame that came with my EOS 5D Mk4, which helps avoid knocking the cable out during use.)

* Apple USB C to Thunderbolt 2 adapter (~ $50):
https://www.apple.com/shop/product/MMEL2AM/A/thunderbolt-3-usb-c-to-thunderbolt-2-adapter
(required for recent Macs that only have USB-C ports)

It works perfectly with my 5D Mk4 and also with a friend's Nikon camera that has an HDMI output. It should basically work with any camera that has HDMI. My Mac is a 2017 MacBook Pro Core i5 3.1Ghz 4x USB-C Touchbar model, but this setup should work with any USB-C Mac. I have found that the HDMI stream does use a bit more CPU than v002 Camera Live Syphon did, so keep that in mind if you're using a less powerful Mac.

Again thanks Tom for all your work on this project. I hope the above info assists anyone who's looking for an alternative.

bangnoise, 2019-11-19:
Please try with [the latest release](https://github.com/v002/v002-Camera-Live/releases/latest)

ddavidebor, 2019-12-01:
Working for me on Mojave and Wirecast

closed 2019-12-02 by bangnoise