v002/v002-Camera-Live #58: Change live view resolution from 1056x704 to 1920x1080

delphiy, 2019-07-19:
1. Connect camera, start Camera Live
2. Download [Simple Client](https://github.com/Syphon/Simple/releases/download/5/Syphon.Simple.Apps.zip)
3. Open Simple Client, your camera should be displayed - if not, select it from the menu in the toolbar
4. Read the dimensions from the status display in Simple Client's toolbar
5. Set those values in CamTwist > Preferences... > General > Video Size > Custom

This is a CamTwist usage issue, please contact them for further support.

_Originally posted by @bangnoise in https://github.com/v002/v002-Camera-Live/issues/44#issuecomment-513199636_

delphiy, 2019-07-19:
Hi,
First of all , thanks 
I can see camera without black bars because I set the correct resolutions in CamTwist to  1056x704. How I set my camera live view to 1920x1080 so I can get 1920x1080 in CamTwist .

Thanks

bangnoise, 2019-07-20:
The camera's resolution is fixed. You can't change it.

**This is a CamTwist usage issue, please contact them for further support.**

closed 2019-07-20 by bangnoise