v002/v002-Camera-Live #61: Stuck on starting (Canon Rebel T3i)

montikid, 2019-10-10:
I've done everything! to make this work with no luck. After I downloaded Camera Live I was able to use it once with CamTwist and recorded directly into QuickTime. After that, every time I've tried to use it, I'm stuck with the "starting" message and my camera highlighted in blue.  I downloaded new firmware, tried plugging and unplugging the camera, tried closing and opening the app - nothing!  Anyone have any idea on a workaround for this issue? I have a MacBook Pro running on Mojave. Terrified to upgrade to Catalina with this ongoing issue.
[image]

simonhaenisch, 2019-11-19:
I've encountered something similar with a 550D (T2i)... the reason is probably that before disconnecting the camera, you did not click somewhere else in the list to deactivate it in Camera Live. The way I worked around it was to 1) get back into the screen that you screenshot, 2) click out of the list, 3) turn off the camera, 4) quit Camera live, 5) turn the camera back on, 6) restart Camera Live. That way the camera didn't auto-start anymore when the program launched, and I was able to get it to activate again.

Another time I started the apps in a different order, which caused an issue with syphon, I think because I started the syphon client (OBS) before the camera was connected to Camera Live (which creates the syphon server)... I was stuck with the same issue that the camera wouldn't switch into "active", and to get it working I had to remove the cyphon client from OBS and add it again.

All of the above was with v9, haven't yet tested with v10.

bangnoise, 2019-12-02:
Please try with the [latest release](https://github.com/v002/v002-Camera-Live/releases/latest).

closed 2019-12-02 by bangnoise