v002/v002-Camera-Live #62: SL2 not working with Mac OSX 10.15.2

ppespepe, 2019-12-12:
Updated to 10.15.2 and cameralive stopped communicating with the cammera.

bangnoise, 2019-12-15:
Unfortunately Canon have confirmed an issue with the Canon SDK and 10.15.2. They promise a fix but no date as yet.

mariohernandez, 2019-12-18:
Anyone has any workarounds while we way for a fix?  can't use my camera at all after latest macos update.

tavon, 2019-12-19:
+1 Same issues here. Where can we get updates on the status of Canon SDK support for Catalina 10.15.2? I searched pretty extensively on their developer forum, but I couldn't find it.

ClearCloud8, 2019-12-24:
Same issue... would love to be able to use this software on Mac OS Catalina.  This software saves me from needing to buy a Cam Link 4k....   I hope this can be fixed soon....  otherwise it's back to live streaming with crappy iSight camera instead of DSLR.  :-|

michiels, 2019-12-29:
For what it's worth, here's a statement on Canon's website indicating that they are aware: https://t.co/KFvp63zFNJ?am

ThomasBallatore, 2020-01-16:
I haven't tried yet, but comments in the link below suggest that the SDK is working again in 10.15.3 beta.

https://www.dpreview.com/forums/thread/4434325?page=2

Anyone willing to try that out and report if Camera Live is working again? I've been sticking with 10.14.6 because of this issue.

sandeld, 2020-01-28:
I just updated my Macbook to the macOS 10.15.3 beta, and it's still not working for me either. I have a Canon T2i with the latest firmware installed. 

I have confirmed that the EOS Utility Live View works, but whenever I launch Camera Live, it either crashes immediately, or it does open up but says there's a camera error. 

So, still not usable. :-(

ThomasBallatore, 2020-01-30:
And...it's working on 10.15.3! So happy to have this old friend back. In the meantime, I did by a Cam Link 4K dongle which is performing well—great to have both options again.

[image]

sandeld, 2020-01-30:
I just updated to 10.15.3 and it's still not working for me. :-(

It crashes immediately after I open it, and then when I click "reopen" it says Camera Error.

[image]
[image]

bangnoise, 2020-01-30:
There's a new Canon SDK release, supposedly with a fix for this issue. Just testing with it now.

bangnoise, 2020-01-30:
Fixed, please update to [the latest version](https://github.com/v002/v002-Camera-Live/releases/latest).

closed 2020-01-30 by bangnoise

jrong1271, 2020-01-30:
🥂 Just want to say Thank you for fixing this, you guys are amazing! 💯 👍

ClearCloud8, 2020-01-31:
It works for me as well!  Thank you so much!!  :)

sandeld, 2020-01-31:
Yessss!! Thank you so much!

terencetyrone, 2020-10-08:
Sony A7ii 
I continually get "camera error" every time I connect to my iMac. Please help!