v002/v002-Camera-Live #64: Stuck in starting mode, never switches to activate

kiarafbickers, 2020-02-05:
Hey there, sorry if this isn't the proper chancel for this issue, but this is what I'm seeing.

In [this video tutorial](https://youtu.be/kMbBCuwHzVA?t=76), I see that I should be able to click on my selected camera and have it go from `Ready` to `Starting` to `Active`. But mine doesn't seem to be clickable it just stays in starting mode.

[image]

Have you seen this before? Is there a setting or something that needs to be changed to make this work?

bangnoise, 2020-02-06:
Hi -

1. Check you are using the [latest release](https://github.com/v002/v002-Camera-Live/releases/latest).
2. Carefully check the [Troubleshooting](https://github.com/v002/v002-Camera-Live#troubleshooting) section of the README.

tavon, 2020-02-27:
Thanks for the latest update. I can confirm that it was working for a couple weeks then for some reason, it's freezing when USB is connected to my 70D and the Camera Live app is stuck in "Application Not Responding". The only way to close it is to Force Quit or kill the process. 

[image]

tavon, 2020-02-27:
Oddly, when I use Ecamm Live which supports EOS cameras like Camera Live, it works. I know this isn't enough information... Sorry for not providing more useful info.
[image]

Deeptradingfx, 2020-03-02:
> Thanks for the latest update. I can confirm that it was working for a couple weeks then for some reason, it's freezing when USB is connected to my 70D and the Camera Live app is stuck in "Application Not Responding". The only way to close it is to Force Quit or kill the process.
> 
> [image]

 restart you Mac and your camera some times needs to change the USB port also b4 you start camera live make sure you live view  is turned on! other thing make sure you EOS utility are not running on start up ...

voik, 2020-03-09:
I'm having the same issue where Camera Live does not get beyond "Starting."  I am running the latest release, Build 11. I have thoroughly reviewed the troubleshooting list. The camera is connected via USB, the camera is a 6D which appears in CAMERAS.md, the camera's firmware has been updated to canon's latest version (1.1.9), EOS utility is not running, and the camera's wifi is set to "Disable." 

I have also put the camera in live view when starting the app. I've also tried rebooting my computer and trying multiple USB ports. I am running macOS Mojave 10.14.6 on a 2014 MacBook Pro.

No dice. Am I missing something?

Deeptradingfx, 2020-03-12:
@voik this is because for some reason you turned off the camera or restarted your pc with the camera connected to the live camera app you always need to close the application and disconnect your camera before restart ou turn off your camera I believe that something related to the addressing of the usb always happens to me what I normally do put the camera in manual mode reset it to factory defaults remove the battery or charger for 3 minutes, than open the camera live app before connecting the usb cable on the camera and make sure the camera is  in live view mode connect the usb cable! if it still stuck on starting mode click on the live view button few times  until start! my one is T3i other thing I downgrade from Catalina to Mojave coz on Catalina the de video streaming was freezening!  sorry for my English

Deeptradingfx, 2020-03-12:
sure its will work for you too @tavon tavon

tavon, 2020-03-17:
Now that I've uninstalled Ecamm Live, Camera Live is working again.

bangnoise, 2020-03-18:
Thanks for the update @tavon - annoying if having Ecamm installed is blocking other applications from using the camera. I'll look further into this and see if we can at the very least provide a meaningful error.

bangnoise, 2020-03-18:
@kiarafbickers would be interested to hear if you have Ecamm Live installed?

GlanR001, 2020-03-19:
I'm having this same issue with an EOS 1200D. I don't have any other apps installed such as Ecamm Live. Anybody have any ideas?

Snooz82, 2020-03-20:
me too.

EOS R (Firmware 1.6.0)

Worked fine the first time. after changing battery never worked again.
just stays in starting mode.

Is there any way we could give you some debug logs, or give any more information?

Cheers

lancetosborne, 2020-03-23:
Same issue here with the Canon M50. Camera Live used to work after a few kills and restarts, but when I heard a new version was available, I decided to try it. I can't get the new version to go "Active" at all.

I do have other camera tools installed (OBS, etc.), but none are running after killing them in Activity Monitor. Interestingly, I noticed that CL is hogging CPU while starting. 

CL is a great little tool. Hope there's a solution soon. Thanks for working on this project.

[image]

oselz, 2020-03-24:
I have this same issue, the workaround I have found is:
- Ensure the camera is connected by USB and turned on
- Ensure no camera is selected in Camera Live (click below the listed cameras to deselect the camera)
- Quit Camera Live
- Start Camera Live (no camera will be selected by default)
- Select your camera

Basically, it seems to work if Camera Live starts and _does not_ auto-connect to the camera - ie manually connecting seems to work fine.

Other notes:
- Quitting Camera Live after getting this starting issue will cause it to crash.
- Connecting or turning the camera on after Camera Live is running will not work
- This is with Build 11
- OS is Catalina
- Camera is EOS 70D

Snooz82, 2020-03-24:
@oselz You really made my day!!!

This work-around does actually work!

Tested with EOS-R

bangnoise, 2020-03-24:
Here's an alpha build which may fix this issue - it uses libgphoto2 intead of the Canon SDK.

https://github.com/v002/v002-Camera-Live/releases/download/12/Camera.Live.zip

jacuvleo, 2020-03-25:
Just downloaded the new alpha build - at first still got "camera error" followed by the opening and closing of the shutter as if it were taking a picture, about every 3 seconds, but no images were stored on the SD from it. messed around with it a bit, unplugged/plugged and switched to manual in no ordered fashion, and then it gave me the "active" message and ended up working

bangnoise, 2020-03-25:
@jacuvleo ah thanks for the feedback - which camera model are you using?

jacuvleo, 2020-03-25:
I'm on a Canon Rebel XS

lancetosborne, 2020-03-25:
> Here's an alpha build which may fix this issue - it uses libgphoto2 intead of the Canon SDK.
> 
> https://github.com/v002/v002-Camera-Live/releases/download/12/Camera.Live.zip

This works for my M50 + Mojave setup. Thank you very much!

sihil, 2020-04-05:
Not tried alpha 12, but alpha 13 resolved the `Starting...` issues I was experiencing. Many thanks for all your efforts @bangnoise