v002/v002-Camera-Live #65: This compared to the HDMI feed

dvcrn, 2020-02-26:
Huge thanks for this app! I've been looking for something like this for a long time. 

I am curious how this video feed compares to the direct HDMI out of the camera, latency ignored. Are there any infos on bitrate, color precision and so on?

Deeptradingfx, 2020-03-02:
> Huge thanks for this app! I've been looking for something like this for a long time.
> 
> I am curious how this video feed compares to the direct HDMI out of the camera, latency ignored. Are there any infos on bitrate, color precision and so on?

Typical latency of a Canon 7D is 120 ms (between 3 and 4 frames at 30 FPS), which is comparable to the latency of the same camera's HDMI output connected to a capture device. please close your issue ...

bangnoise, 2020-03-02:
@Deeptradingfx not a helpful comment

bangnoise, 2020-03-02:
One big difference for some models the HDMI output has the UI superimposed, whereas the live-view feed doesn't. As far as I'm aware, they are both always derived from the live-view (rather than direct sensor data).

The live-view feed via USB is encoded as a jpeg - iirc it comes with no data about color, and I haven't done a survey of image quality across different models.

closed 2020-03-02 by bangnoise

Manouchehri, 2020-04-11:
@bangnoise Is the resolution on HDMI also the same as the live-view feed?

Manouchehri, 2020-04-28:
Answer to myself, no, HDMI on my Canon M100 is 1080p! =D