v002/v002-Camera-Live #66: Canon T3i catalina video Freezes

Deeptradingfx, 2020-03-02:
im using the last version of camera live on Catalina 10.15.3 reconized ok, start ok but when get the video feed its runs for 3 seconds and stack its working normally on Ecamm live! than its mean usb is ok every thing is ok but camera live not working

hershal, 2020-03-19:
I'm having the same issue with a T1i. Video will come up for 3-5 seconds and then freeze until I restart Camera Live. Is there anything from us that you need to help fix or replicate the issue?

bangnoise, 2020-03-19:
There are reports of Ecamm Live preventing other applications accessing the camera - a user found Camera Live worked for them after uninstalling Ecamm Live.

hershal, 2020-03-19:
I don't have Ecamm Live installed -- I didn't even know what that was until I looked over the other issues

bangnoise, 2020-03-19:
@hershal OK, well you're not having the same issue

hershal, 2020-03-19:
Oh I see. Same symptoms, different issue.

I tried it again after a reboot and it's more sporadic. Sometimes I can get a video feed lasting about a minute, sometimes about 3 seconds. I haven't been able to find commonalities yet

bangnoise, 2020-03-19:
🤷🏻‍♂️

@hershal if you are able to, the following information may help:

1. get v002 Camera Live into the "stuck" state
2. open Activity Monitor (/Applications/Utilities/Activity Monitor)
3. select Camera Live in the list
4. From the cog on the toolbar, select "Sample Process"
5. save the resulting file using the "Save..." button
6. attach the file to this issue (drag and drop)

Deeptradingfx, 2020-03-19:
> Oh I see. Same symptoms, different issue.
> 
> I tried it again after a reboot and it's more sporadic. Sometimes I can get a video feed lasting about a minute, sometimes about 3 seconds. I haven't been able to find commonalities yet
this  issue  is only on catalina i downgrade to mojave and working great! thats was the only solution  i found after  2 weeks trying to fix it on catalina... was working fine on Ecamm live i dont know how they done great job on ecamm live!  how ever its been done base on camera live app ....

bangnoise, 2020-03-19:
@Deeptradingfx please stop commenting on every single issue here

hershal, 2020-03-19:
Attached. I also noticed that CPU usage spikes to 100% when it freezes.

[Sample of Camera Live.txt](https://github.com/v002/v002-Camera-Live/files/4356970/Sample.of.Camera.Live.txt)

hershal, 2020-03-19:
Note that I have a hackintosh so if you see some weird stuff (MacPro with a Core i5 and AMD 7970), it's because this is a custom PC running Catalina.

bangnoise, 2020-03-19:
Thanks - just in case, I'd try disconnecting any other USB devices. If that doesn't help I suspect this is another issue inside the Canon SDK and another reason for a complete rewrite using an alternative library...

bangnoise, 2020-03-19:
@Deeptradingfx oops very sorry, too many issues on the go - this one is your issue, commenting not unreasonable! Are you on a hackintosh too?

dantaylorseo, 2020-03-23:
Following - having same issues with 800d

bangnoise, 2020-03-23:
Currently working on a rewrite to avoid the Canon SDK entirely - will post a build as soon as I have it

dantaylorseo, 2020-03-23:
One note, this works perfectly with OBS, when I select a syphon source.

When I try and use camtwist it freezes 3-4 seconds later and I need to force quit camera live and use another usb port. Just info to see if it helps.

Whats the ETA on the new version? Days, weeks, months? Just so we know :)

bangnoise, 2020-03-23:
Oh interesting - @hershal and @Deeptradingfx it would be good to know what app you are each Syphon-ing to.

dantaylorseo, 2020-03-23:
@bangnoise let me know if you need any logs or to jump on a screenshare

hershal, 2020-03-23:
I’m using CamTwist but I can try out OBS tonight

Deeptradingfx, 2020-03-23:
> @Deeptradingfx oops very sorry, too many issues on the go - this one is your issue, commenting not unreasonable! Are you on a hackintosh too?

No problem  thanks for your great work on this Project! yes im on hackintosh too but also have a macbook only on catalina  obs and wirecast  Video will come up for 3-5 seconds and then freeze until I restart i Camera Live i on usb 3.0 downgrade to mojave on hackintosh and its working again as allways been! on usb 2.0 on catalina hackintosh stay up for 1 or 2 minutos b4 freezer and some times stuck on starting

dantaylorseo, 2020-03-23:
Some more info on this. I notice it happens more if I move the camera around... not sure if this is coincidence?

Also the CPU runs at around 18% while running fine, with just 1% GPU usage, when it freezes the CPU pins to 100% until force quit.

bangnoise, 2020-03-23:
@dantaylorseo perhaps your problem is being exacerbated by a loose cable? Try with another USB cable if you have one.

dantaylorseo, 2020-03-23:
Already tried multiple cables including a brand new one. Where can I get the Canon SDK and then I can run in Xcode and send you debug?

hershal, 2020-03-23:
I also notice the same thing @dantaylorseo mentioned. If I leave the camera still then it seems to be ok for a while, but if I wave my hand in front of it or move the camera then the stream freezes. If I connect/disconnect my camera at this point then Camera Live crashes. It freezes in both OBS and CamTwist.

hershal, 2020-03-23:
FYI I'm using the same cable that I used when Camera Live worked under High Sierra

dantaylorseo, 2020-03-24:
@bangnoise it seems the OBS thing was an anomaly. I just had it running perfectly in MS Teams as a webcam for 15 minutes now it's back down to 2-3 seconds.

Also I don't think its your software, as you have suggested before it's the EOS SDK. The EOS utility does exactly the same when I try live view shoot. It freezes the display after 2-3 seconds and CPU peaks to 209%!

I think your rewrite might be the best solution.

bangnoise, 2020-03-24:
I'd be grateful if anyone could try this version - it uses libgphoto2 intead of the Canon SDK.

Feedback welcome - ideally in a new GitHub issue, not this one.

https://github.com/v002/v002-Camera-Live/releases/download/12/Camera.Live.zip

dantaylorseo, 2020-03-24:
Will do now

bangnoise, 2020-03-26:
New alpha version - version 12 didn't work on older os versions
https://github.com/v002/v002-Camera-Live/releases/download/13/Camera.Live.zip

Deeptradingfx, 2020-03-30:
> New alpha version - version 12 didn't work on older os versions
> https://github.com/v002/v002-Camera-Live/releases/download/13/Camera.Live.zip

you are the best man libgphoto2 working so great! closing 5 starts to you

closed 2020-03-30 by bangnoise

luruke, 2020-08-05:
I have very similar issue to @dantaylorseo, running Catalina on Hackintosh.
EOS utility and official canon apps crashes after 2-3 seconds.
v13 using libgphoto works great, but quality and framerate is very poor