v002/v002-Camera-Live #67: Canon EOS Digital Rebel XT "Camera Error"

kennyrho, 2020-03-12:
I have looked far and wide with no results on this issue for my particular model. 

Has anyone experienced a "Camera Error" on the Camera Live Mac app for a Canon EOS Digital Rebel XT? I am running Catalina 10.15.3. I am able to transfer data and my computer recognizes the device on the EOS app (version 2). 

Is someone able to provide some troubleshooting tips?

Thank you everyone,

LAVD

bangnoise, 2020-03-19:
Do you have Ecamm Live installed? Some users have reported a conflict where Ecamm Live prevents other applications using the camera.

Commaimagination, 2020-03-21:
Same problem with my Rebel XTi, My 7D work super fine. I guess that the problem might be caused by the fact the the Xti has no movie mode. (the 7D has movie mode)

bangnoise, 2020-03-21:
Unfortunately the Rebel XT and XTi are not on Canon's list of supported cameras.

closed 2020-03-21 by bangnoise