v002/v002-Camera-Live #68: Is it possible to adjust camera settings like aperture using Camera-Live and CamTwist?

ThSGM, 2020-03-15:
I've not been able to figure out how to change the aperture settings on a T3i while hooked up to Camera-Live and CamTwist. I have tried to change the camera settings before linking it to the USB (setting either manual or Av) but the aperture always seems to go to wide open.

bangnoise, 2020-03-15:
This depends on the camera model. v002 Camera Live receives the LiveView image. On a 7D, the following works:

1. Turn the photo/video toggle to photo mode
2. Put the camera into manual (M)
3. Turn on the display on the camera (using the Start/Stop button) (the camera rejects setting changes while USB is connected but the display is off)
4. Make any adjustments to aperture, etc you want
5. If you wish, you can disable the display once you have the settings you want

trinh, 2020-03-16:
> This depends on the camera model. v002 Camera Live receives the LiveView image. On a 7D, the following works:
>
> 1. Turn the photo/video toggle to photo mode
> 2. Put the camera into manual (M)
> 3. Turn [...]

Hi, 

You are absolutely right. I re-tested and really put the aperture high on manual (rendering the image darker and darker) and that was enough to confirm that the settings were working!

This issue can now be closed.

ThSGM, 2020-03-16:
Hi,

You are absolutely right. I re-tested and really put the aperture high on manual (rendering the image darker and darker) and that was enough to confirm that the settings were working!

This issue can now be closed.

closed 2020-03-16 by ThSGM