v002/v002-Camera-Live #69: Canon 80D "No camera"

dcajuste, 2020-03-21:
Hello peeps!

I am super new to this and I hope I am not annoying with this potentially simple question. I am using an Elgato h60S Capture to use my Canon as a Webcam. I was guided to download Camera Live and then Cam twist. However, When plugging in the Camera for Camera Live nothing comes up. Period. 

I have used Game Capture HD from Elgato and I go get a signal. So I am not sure if its just with Camera Live? I have used two different computers to test this out. Still nothing.

I really appreciate your time and patience.

Best,
D

bangnoise, 2020-03-24:
The Elgato is for HDMI capture (from your camera's HDMI output). Camera Live is for USB capture from the USB connection.

Connect the camera via USB.

Yes there are a lot of problems with the official Canon SDK. I've just put out an alpha build using an alternative library - you may want to try it instead - https://github.com/v002/v002-Camera-Live/releases/download/12/Camera.Live.zip

bangnoise, 2020-03-28:
Closing this issue as it sounds like you were connecting via HDMI.

closed 2020-03-28 by bangnoise