v002/v002-Camera-Live #70: Camera Live Not finding 5D Mark ii

ckopix, 2020-03-23:
Camera Live works great with my 5D Mark ii on Macbook Pro w/Catalina.
However, on my Mac Mini with Catalina it will not find the camera.  Camera connection is fine as EOS utility sees it and can control/view video.  I have no other apps running.  I haver rebooted, tried different USB ports, etc.  Can't seem to figure out why Camera LIve is not seeing it on the Mac Mini.
[image]

bangnoise, 2020-03-23:
Very strange - I'm in the middle of a complete rewrite using libgphoto2 instead of the official Canon SDK. I'll link a build here as soon as I have one for you to test.

ckopix, 2020-03-24:
I have no idea why but camera live suddenly is able to see the 5D Mark II.

closed 2020-03-24 by ckopix

bangnoise, 2020-03-24:
If you have problems with it again, try the alpha build I just put up - https://github.com/v002/v002-Camera-Live/releases/download/12/Camera.Live.zip