v002/v002-Camera-Live #71: 5D Mark III not showing up

sarobar, 2020-03-24:
I am not seeing 5D Mark III. Is there a instruction on what mode the camera should be when connected? I tried to turn in on and its on Manual and live mode.

bangnoise, 2020-03-24:
There are a load of problems with the Canon SDK. Here's an alpha version based on an alternative library (libgphoto2). Might help your problem.

https://github.com/v002/v002-Camera-Live/releases/download/12/Camera.Live.zip

jamputch, 2020-03-31:
I'm having the same issue. Tried with the version above but still no luck.

fpo4now, 2020-03-31:
Does shutting off the System Integrity Protection have an effect? 
https://www.imore.com/how-turn-system-integrity-protection-macos

kellerchch, 2020-03-31:
I am trying to use these instructions: https://docs.crowdcast.io/en/articles/1935406-how-to-use-your-dslr-as-a-webcam to capture video with camtasia as my recording/editing solution. When I load the video successfully in the camtasia recorder, the error I get when I 'start recording' is 

Waiting for device initialization
The device is taking longer than expected to initialize. Check your device.

I am grateful you @bangnoise have created this cool software. I'm not a developer and I'm running a company on a shoe string but I applaud you and thank you and am wondering if there is something more I can do. Perhaps TechSmith, the company that made Camtasia would be willing to fund your development of a solution. What you are providing is so critically valuable in a burgeoning market space of small individuals creating video content.

jamputch, 2020-03-31:
I tried a few different cables and found that some worked and others didn't. Don't have the time to troubleshoot any further but perhaps cable quality or length?