v002/v002-Camera-Live #73: New Version (12) not working on macOS 10.13

dcartman, 2020-03-26:
Trying the new version (12) on macOS 10.13. Below is the error I see in terminal when running, seems like a linking issue 10.15 vs 10.13.

dyld: lazy symbol binding failed: Symbol not found: ____chkstk_darwin
  Referenced from: /Users/daniel/Downloads/Camera Live.app/Contents/MacOS/./../Frameworks/libgphoto2.6.dylib (which was built for Mac OS X 10.15)
  Expected in: /usr/lib/libSystem.B.dylib

dyld: Symbol not found: ____chkstk_darwin
  Referenced from: /Users/daniel/Downloads/Camera Live.app/Contents/MacOS/./../Frameworks/libgphoto2.6.dylib (which was built for Mac OS X 10.15)
  Expected in: /usr/lib/libSystem.B.dylib

Abort trap: 6

bangnoise, 2020-03-26:
Sorry about that - fixed in [build 13](https://github.com/v002/v002-Camera-Live/releases/download/13/Camera.Live.zip)

closed 2020-03-26 by bangnoise

dcartman, 2020-03-26:
Thank you for the quick fix.