v002/v002-Camera-Live #74: PowerShot G7X Mark ii

fmorris, 2020-03-26:
Hi! I get a "camera error" when I try to connect my PowerShot GX7 Mark ii. My old 500D works just fine. Any suggestions? I have the newest firmware on the GX7. Kindly regards Fredrik

bangnoise, 2020-03-26:
Which version of Camera Live and macOS?

bangnoise, 2020-03-26:
For version 11, the Canon SDK doesn't list the Mark II as supported (just the Mark III). 
For the alpha version 12 it may work.

fmorris, 2020-03-26:
Its Catalina 10.15.3 and I have tried both 11 and 12 alpha. Will it be added in the future?

benistary, 2020-04-19:
I've tried 11 stable and 12/13 alpha versions and it's still shows "Camera Error" message. (Catalina 10.15.4)

Borai, 2020-04-23:
Same here. Tried 11 stable and 12/13 alpha with the latest Catalina.

ourarash, 2020-05-07:
Even if Camera Live works, PowerShot G7X Mark ii has the main problem that the lens doesn't even open when you connect it to USB.