v002/v002-Camera-Live #75: Stuttery Video

dcartman, 2020-03-26:
First, thanks for building this software, I truly appreciate it. I have a Canon T5i, and love it. In the time of COVID, I'm excited to get started with it as a Web Cam.

Ok, so the issue.

The video I'm receiving from the camera is stutters badly, and feels laggy dropped frames, etc.

At first I thought it was the computer I was using, 2011 Mac Book Air, but when connecting it to newer Macs with way more horse power I'm seeing the same issue. 

So first question, should I be expecting smooth video?

Second question, is there anything I can do to improve the experience? Either on the camera or with the software?

bangnoise, 2020-03-26:
Which version of Camera Live are you using? The "stable" 11 or an alpha ([13](https://github.com/v002/v002-Camera-Live/releases/download/13/Camera.Live.zip) is the latest).

dcartman, 2020-03-26:
I've tried both. I started with 11, and hoping that it would resolve the stuttering I tried 13 this morning.

bangnoise, 2020-03-26:
Anything else plugged in to USB?

What app are you Syphoning to?

vade, 2020-03-26:
Do you happen to have the shutter speed set to anything other than 1/30 or 1/60th? some camera's allow low shutter speed which results in stuttery looking motion if you are on some modes. Shot in the proverbial dark.

dcartman, 2020-03-26:
@bangnoise - 

Anything else plugged in to USB: When testing this morning I only had the Camera plugged in.

What app are you Syphoning to: OBS, and I have also tried CamTwist. Both had the same issue.

@vade - I set the camera to 1280 60fps. It was 1080 24fps. Still jittery and laggy.

bangnoise, 2020-03-26:
Try putting the camera into photo mode, not movie mode.

bangnoise, 2020-03-26:
also try another USB cable - as short as possible, to eliminate that as a possibility.

dcartman, 2020-03-26:
Well, I've tried three different cables all with the same result.

I tried all the video frame rates and size, no luck there. I also set the picture capture quality to s1 instead of RAW thinking that might be the issue, but still no luck.

The camera is connecting at USB2 speeds, which is all the camera supports. 

If I build this locally, is there a good place to attach and monitor buffers/traffic coming from the camera?

deantaylor, 2020-03-27:
I'm also having this problem. Latest stable build (11) with Canon 650D. Using in OBS.

dcartman, 2020-03-28:
@deantaylor What does your setup look like? (camera settings, cable, computer, software)

deantaylor, 2020-03-28:
@dcartman I'm using 2012 13" MacBook (this may be the issue). 30fps, camera mode - Camera Live > OBS. Standard USB Cable

dcartman, 2020-03-28:
@deantaylor I figured it was my 2011 Mac Book Air, but I've since tried it with much newer Macs and have the same issue. 

I've gone back and tried builds 8 - 13 at this point.

13 is by far the best of the lot (still jittery with frame drops). Also the CPU and memory performance is remarkable compared to the earlier builds. CPU usage is down in the 20% range, was over 100, memory is around 28M where it was closer to 75 or 80.

jhonathanmerkas, 2020-03-31:
Some solution happens to me the same

bangnoise, 2020-04-01:
Does it make a difference if Camera Live is the front-most app?

deantaylor, 2020-04-01:
@bangnoise  I've set this up and tried. Unfortunately, no difference. You don't think this can be something to do with the age of the computer?

bangnoise, 2020-04-01:
@deantaylor see above where @dcartman said:

> @deantaylor I figured it was my 2011 Mac Book Air, but I've since tried it with much newer Macs and have the same issue.

When it's running happily, my Canon 7D gives somewhere around 20 FPS - it's not a steady rate but it's good enough with no visible stuttering.

Generally, some people have had success with the following:

- use a shorter USB cable
- disconnect other USB devices
- don't run USB cables alongside power cables or other sources of interference
- if you own a mains power adapter for the camera, use that instead of the camera's batteries (I wouldn't buy one to solve this because it may not help)

If you haven't already, try both the "stable" [current release](https://github.com/v002/v002-Camera-Live/releases/latest) and the [alpha build 13](https://github.com/v002/v002-Camera-Live/releases/download/13/Camera.Live.zip).

Note for myself - cameras in this issue are T5i, T5 and 650D.

pointmax, 2020-04-10:
Similar issue here. on my 13" 2018 MacBook pro.
I also tried the official eos utility live view, just to see if it's a usb problem. The video feed from the EOS utility is very smooth, but the app consumes a lot of cpu power and makes my macbook sound like a jet engine. So I prefer stuttering video with super low cpu consumption.
I tried the 11 and 13 versions with the EOS 600D (rebel t3i), both had very similar video output.
The video feed is smoother when I set the camera to photo mode (with a manual shutter speed of 1/60) instead of video mode; however, the display turns off when not in video mode.
Overall I am really satisfied with this nifty little app, I wish I could use it with my Sony one day…

sters, 2020-04-20:
I encountered the same issue as a lower fps about 5-10fps (my feeling, not collect frames). On EOS Kiss x7i (with EF-S18-55mm, almost using 18-25mm range), 13" 2018 MacBook pro, CameraLive Version 13.

I tried the change camera setting as same as @pointmax said solution, then it was resolved.
Set photo mode with the Tv (prioritize shutter speed) and set 1/60 shutter speed, And set 1280x720 30fps on CamTwist preference.
If use Av mode (prioritize aperture) with F4, it's not smoothing. 

Also, It's not works on video mode. I think it's correctly.

ithinkimongithub, 2020-08-12:
I have an EOS M50 and I get terrible stutter (it just freezes for 5 seconds about every 15 seconds). Based upon my camera, opening up the camera's own lcd display eliminates this stutter. I'm in Tv mode but there's only 1280x720 available for this camera on whichever utility I have.