v002/v002-Camera-Live #79: Camera Support Request: Canon EOS M6

paazel, 2020-03-27:
Hello, wanted to put a request in for the Canon EOS M6, I've tried it with Version 11, and it does not get past "Starting"

Unrelated, it works fine with another camera that is listed on the supported camera list.  Thanks for your fine work!

bangnoise, 2020-03-27:
Hi - have you tried with the [version 13 alpha](https://github.com/v002/v002-Camera-Live/releases/download/13/Camera.Live.zip)?

Let me know if it works, it's useful to track success for different models.

paazel, 2020-03-27:
Yes, I tried both 12 and 13 without success.

msgpacket, 2020-03-31:
I have also tried my M6 without success. Because even EOS Utility does not support tethered USB shooting for the M6 it may be gimped in this regard.  People on BackyardEOS are reporting similar results.

bangnoise, 2020-03-31:
Yes the Canon SDK (used in version 11) doesn't support the M6. As it's not supported by Canon I'd guess it won't work with libgphoto2 (versions 12 and 13) either - once we've confirmed that in a future build I'll add a useful error.

ijmarshall, 2020-04-12:
I have tried with my Canon EOS M6 (version 1) and get an identical error with the version 13 alpha.
If this helps...  The M6 is listed as supported on libgphoto2 for liveview: http://www.gphoto.org/proj/libgphoto2/support.php
I have also successfully got a liveview (albeit with 1 second latency) using the gphoto2 CLI together with ffmpeg and ffplay on my Mac using the M6.

bangnoise, 2020-04-12:
@ijmarshall good to know it works with gphoto2, means I'll be able to get it going in Camera Live too

paazel, 2020-04-12:
@ijmarshall @bangnoise WOW! That would be amazing to see this camera work! The STM lenses and Dual Pixel Auto Focus makes this a game changer for recording in the home!

shawnrif, 2020-04-17:
just vocally SECONDING the request for (mk1) compatibility. it's the best-cheapest camera i've found, and would rather not spend more on the M50 or M6mk2 just for camera-live compatibility. 

btw, thanks for this work, it's amazing.

msgpacket, 2020-04-18:
Nice to be proven wrong. Good luck with the effort!

msgpacket, 2020-04-30:
> I have tried with my Canon EOS M6 (Mark 1) and get an identical error with the version 13 alpha.
> If this helps... The M6 is listed as supported on libgphoto2 for liveview: http://www.gphoto.org/proj/libgphoto2/support.php
> I have also successfully got a liveview (albeit with 1 second latency) using the gphoto2 CLI together with ffmpeg and ffplay on my Mac using the M6.

@ijmarshall I think I have the dependencies installed on my MacBook, can you please tell me what gphoto2 command line you used?

bangnoise, 2020-04-30:
@ijmarshall or @msgpacket - I wonder if you could run the following with your M6 attached and attach the results to this issue? Thanks!

Record the output of `LANG=C gphoto2 --summary >summary.txt` to get generic summary information into the summary.txt file.
Record the output of `LANG=C gphoto2 --list-all-config >config.txt` to get the configuration tree into the config.txt file.

msgpacket, 2020-05-02:
@bangnoise Here you go

[summary.txt](https://github.com/v002/v002-Camera-Live/files/4567524/summary.txt)
[config.txt](https://github.com/v002/v002-Camera-Live/files/4567525/config.txt)

MrJ-L, 2020-05-27:
I'd also love for this to work somehow!

MrJ-L, 2020-07-29:
I just tried the v13 alpha and it recognises the camera but I get the 'Camera Error' warning when I try to use it. Does anyone have a way around this yet?

bendonaldson01, 2020-10-19:
Has anyone been able to get it functioning?