v002/v002-Camera-Live #85: Camera Live hangs at 'Starting' or results in 'Camera error'

hwschuur, 2020-04-01:
**Describe the bug**
I launch the app with my camera (a Canon 40D) off. When I turn my camera on, the camera is recognised as a Canon EOS 40D (PTP mode) and Camera Live has status 'Ready'. When I (double) click the camera, Camera Live status changes to 'Starting' and I hear the 'Live view' mode of the camera being initialised (click sound). Then, Camera Live hangs at the 'Starting' status, or the status changes to 'Camera Error'.

**To Reproduce**
Steps to reproduce the behavior:
1. Turn off camera
2. Launch Camera Live
3. Turn on camera
4. Double click the camera in Camera Live
5. Camera Live hangs or shows 'Camera Error'

**Screenshots**
[image]
**Desktop (please complete the following information):**
 - macOS version: 10.15.4
 - Camera manufacturer and model: Canon 40D
 - Camera Live version: 13 (latest alpha)
**Additional context**
Camera Live has worked quite flawlessly for me in combination with CamTwist when I was still on 10.15.3.

hwschuur, 2020-04-01:
Browsing through the issues, the symptoms I experience seem to be quite similar to those mentioned in issue https://github.com/v002/v002-Camera-Live/issues/64. Might be the same root cause.

hwschuur, 2020-04-01:
Here's some lines Camera Live build 13 writes to the Console:

[image]

hwschuur, 2020-04-01:
Here's some lines Camera Live build 13 writes to the Console (camera disconnected, launch Camera Live, connect camera via USB):
```
default	11:59:08.280134+0200	Camera Live	FRONTLOGGING: version 1
default	11:59:08.280150+0200	Camera Live	Registering, pid=1218
default	11:59:08.281430+0200	Camera Live	CHECKIN: pid=1218
default	11:59:08.290780+0200	Camera Live	CHECKEDIN: pid=1218 asn=0x0-0x71071 foreground=1
default	11:59:08.307288+0200	Camera Live	Registered, pid=1218 ASN=0x0,0x71071
default	11:59:08.307445+0200	Camera Live	Registered, pid=1218 cgConnectionID=a2ea3
default	11:59:08.308734+0200	Camera Live	BringForward: pid=1218 asn=0x0-0x71071 bringForward=1 foreground=1 uiElement=0 launchedByLS=1 modifiersCount=1 allDisabled=0
default	11:59:08.308801+0200	Camera Live	BringFrontModifier: pid=1218 asn=0x0-0x71071 Modifier 0 hideAfter=0 hideOthers=0 dontMakeFrontmost=0 mouseDown=0/0 seed=0/0
default	11:59:08.308871+0200	Camera Live	BringForward: pid=1218 asn=0x0-0x71071
default	11:59:08.308937+0200	Camera Live	SetFrontProcess: asn=0x0-0x71071 options=0
default	11:59:08.314977+0200	Camera Live	Current system appearance, (HLTB: 2), (SLS: 1)
default	11:59:08.318748+0200	Camera Live	Post-registration system appearance: (HLTB: 2)
default	11:59:08.341747+0200	Camera Live	NSApp cache appearance:
-NSRequiresAquaSystemAppearance: 0
-appearance: (null)
-effectiveAppearance: <NSCompositeAppearance: 0x600000569100
 (
    "<NSDarkAquaAppearance: 0x600000568f80>",
    "<NSSystemAppearance: 0x600000569000>"
)>
default	11:59:08.405852+0200	Camera Live	Registering for test daemon availability notify post.
default	11:59:08.405986+0200	Camera Live	notify_get_state check indicated test daemon not ready.
default	11:59:08.467483+0200	Camera Live	SignalReady: pid=1218 asn=0x0-0x71071
default	11:59:08.468178+0200	Camera Live	SIGNAL: pid=1218 asn=0x0x-0x462961
default	11:59:17.735413+0200	Camera Live	Received configuration update from daemon (initial)
default	11:59:18.418611+0200	Camera Live	LSExceptions shared instance invalidated for timeout.
error	11:59:19.464323+0200	Camera Live	gp_camera_capture_preview: -7 I/O problem```

hwschuur, 2020-04-01:
Here's another one, where Camera Live shorty went from 'Starting' to 'Active' and then to 'Camera Error' again:

```
default	16:25:02.807229+0200	Camera Live	FRONTLOGGING: version 1
default	16:25:02.807255+0200	Camera Live	Registering, pid=9427
default	16:25:02.808662+0200	Camera Live	CHECKIN: pid=9427
default	16:25:02.819397+0200	Camera Live	CHECKEDIN: pid=9427 asn=0x0-0xd10d1 foreground=1
default	16:25:02.834834+0200	Camera Live	Registered, pid=9427 ASN=0x0,0xd10d1
default	16:25:02.835012+0200	Camera Live	Registered, pid=9427 cgConnectionID=19b72b
default	16:25:02.836590+0200	Camera Live	BringForward: pid=9427 asn=0x0-0xd10d1 bringForward=1 foreground=1 uiElement=0 launchedByLS=1 modifiersCount=1 allDisabled=0
default	16:25:02.836668+0200	Camera Live	BringFrontModifier: pid=9427 asn=0x0-0xd10d1 Modifier 0 hideAfter=0 hideOthers=0 dontMakeFrontmost=0 mouseDown=0/0 seed=0/0
default	16:25:02.836738+0200	Camera Live	BringForward: pid=9427 asn=0x0-0xd10d1
default	16:25:02.836810+0200	Camera Live	SetFrontProcess: asn=0x0-0xd10d1 options=0
default	16:25:02.845328+0200	Camera Live	Current system appearance, (HLTB: 2), (SLS: 1)
default	16:25:02.850043+0200	Camera Live	Post-registration system appearance: (HLTB: 2)
default	16:25:02.881649+0200	Camera Live	NSApp cache appearance:
-NSRequiresAquaSystemAppearance: 0
-appearance: (null)
-effectiveAppearance: <NSCompositeAppearance: 0x6000005e0a00
 (
    "<NSDarkAquaAppearance: 0x6000005e0880>",
    "<NSSystemAppearance: 0x6000005e0900>"
)>
default	16:25:02.969954+0200	Camera Live	Registering for test daemon availability notify post.
default	16:25:02.970129+0200	Camera Live	notify_get_state check indicated test daemon not ready.
default	16:25:03.053262+0200	Camera Live	SignalReady: pid=9427 asn=0x0-0xd10d1
default	16:25:03.054301+0200	Camera Live	SIGNAL: pid=9427 asn=0x0x-0x856273
default	16:25:12.479422+0200	Camera Live	Received configuration update from daemon (initial)
default	16:25:12.986112+0200	Camera Live	LSExceptions shared instance invalidated for timeout.
error	16:25:20.759459+0200	Camera Live	gp_camera_capture_preview: -7 I/O problem
error	16:25:20.759991+0200	Camera Live	gp_camera_capture_preview: -1 Unspecified error
error	16:25:20.760384+0200	Camera Live	gp_camera_capture_preview: -1 Unspecified error
error	16:25:20.760771+0200	Camera Live	gp_camera_capture_preview: -1 Unspecified error
error	16:25:20.761053+0200	Camera Live	gp_camera_capture_preview: -1 Unspecified error
error	16:25:20.761415+0200	Camera Live	gp_camera_capture_preview: -1 Unspecified error
error	16:25:22.590054+0200	Camera Live	gp_camera_capture_preview: -1 Unspecified error
error	16:25:22.590349+0200	Camera Live	gp_camera_capture_preview: -1 Unspecified error
error	16:25:22.590657+0200	Camera Live	gp_camera_capture_preview: -1 Unspecified error
error	16:25:22.591042+0200	Camera Live	gp_camera_capture_preview: -1 Unspecified error
error	16:25:22.591453+0200	Camera Live	gp_camera_capture_preview: -1 Unspecified error
error	16:25:22.591801+0200	Camera Live	gp_camera_capture_preview: -1 Unspecified error
error	16:25:27.793133+0200	Camera Live	libgphoto2: <private>
error	16:25:27.797867+0200	Camera Live	Couldn't open camera: -10 Timeout reading from or writing to the port
```

IngmarStein, 2020-04-08:
I get the same error with version 13 and a EOS M5 (version 11 just hangs):
```
default	08:52:46.955993+0200	Camera Live	FRONTLOGGING: version 1
default	08:52:46.956027+0200	Camera Live	Registering, pid=99250
default	08:52:46.957143+0200	Camera Live	CHECKIN: pid=99250
default	08:52:46.967092+0200	Camera Live	CHECKEDIN: pid=99250 asn=0x0-0x561561 foreground=1
default	08:52:46.979499+0200	Camera Live	Registered, pid=99250 ASN=0x0,0x561561
default	08:52:46.979619+0200	Camera Live	Registered, pid=99250 cgConnectionID=5206f
default	08:52:46.980961+0200	Camera Live	BringForward: pid=99250 asn=0x0-0x561561 bringForward=1 foreground=1 uiElement=0 launchedByLS=1 modifiersCount=1 allDisabled=0
default	08:52:46.981013+0200	Camera Live	BringFrontModifier: pid=99250 asn=0x0-0x561561 Modifier 0 hideAfter=0 hideOthers=0 dontMakeFrontmost=0 mouseDown=0/0 seed=0/0
default	08:52:46.981056+0200	Camera Live	BringForward: pid=99250 asn=0x0-0x561561
default	08:52:46.981121+0200	Camera Live	SetFrontProcess: asn=0x0-0x561561 options=0
default	08:52:46.988051+0200	Camera Live	Current system appearance, (HLTB: 2), (SLS: 1)
default	08:52:46.991786+0200	Camera Live	Post-registration system appearance: (HLTB: 2)
default	08:52:47.040556+0200	Camera Live	NSApp cache appearance:
-NSRequiresAquaSystemAppearance: 0
-appearance: (null)
-effectiveAppearance: <NSCompositeAppearance: 0x6000018ada00
 (
    "<NSDarkAquaAppearance: 0x6000018ad880>",
    "<NSSystemAppearance: 0x6000018ad900>"
)>
default	08:52:47.099466+0200	Camera Live	Registering for test daemon availability notify post.
default	08:52:47.099608+0200	Camera Live	notify_get_state check indicated test daemon not ready.
default	08:52:47.163815+0200	Camera Live	SignalReady: pid=99250 asn=0x0-0x561561
default	08:52:47.164689+0200	Camera Live	SIGNAL: pid=99250 asn=0x0x-0x5641569
default	08:52:50.389636+0200	Camera Live	MessageTracer: Falling back to default whitelist
default	08:52:50.390151+0200	Camera Live	Received configuration update from daemon (initial)
error	08:52:55.168941+0200	Camera Live	gp_camera_capture_preview: -1 Unspecified error
default	08:52:57.143079+0200	Camera Live	LSExceptions shared instance invalidated for timeout.
error	08:52:58.187182+0200	Camera Live	gp_camera_capture_preview: -1 Unspecified error
error	08:53:01.215142+0200	Camera Live	gp_camera_capture_preview: -1 Unspecified error
error	08:53:04.228635+0200	Camera Live	gp_camera_capture_preview: -1 Unspecified error
```