v002/v002-Camera-Live #86: Camera Live crash with 'Camera Error'

DMCTowns, 2020-04-01:
**Describe the bug**
Occasionally, during recording/streaming, Camera Live will report a 'Camera Error' and stop working,  sometimes causing the app to crash. Restarting Camera Live resolves the problem with no other action required (i.e. no need to reconnect the camera).

**To Reproduce**
This is hard to reproduce as it's not consistent, but has occurred for me after around 20 - 30 mins of recording

**Expected behavior**
For the app to function without error

**Screenshots**
No screenshots, but a crash report below.

**Desktop (please complete the following information):**
 - macOS version: 10.15.3
 - Camera manufacturer and model: Canon EOS 80D
 - Camera Live version: 11

**Additional context**
Crash report:

Process:               Camera Live [1165]
Path:                  /Applications/Camera Live.app/Contents/MacOS/Camera Live
Identifier:            Camera Live
Version:               11 (11)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           Camera Live [1165]
User ID:               501

Date/Time:             2020-03-31 18:47:15.695 +0100
OS Version:            Mac OS X 10.15.3 (19D76)
Report Version:        12
Anonymous UUID:        83210E5C-746D-7393-D8C0-008AFF6BA60A

Sleep/Wake UUID:       730EC50A-E0E7-41E3-94F7-8C2B9F018AC3

Time Awake Since Boot: 2800000 seconds
Time Since Wake:       77000 seconds

System Integrity Protection: enabled

Crashed Thread:        6  Dispatch queue: info.v002.Camera-Live.Canon.liveview

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000058
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [1165]

VM Regions Near 0x58:
--> 
    __TEXT                 0000000108c8a000-0000000108c95000 [   44K] r-x/r-x SM=COW  /Applications/Camera Live.app/Contents/MacOS/Camera Live

Thread 0:: Dispatch queue: com.apple.main-thread
0   libobjc.A.dylib               	0x00007fff6b0fdd64 _objc_rootAlloc + 17
1   com.canon.edsdk               	0x0000000108de51b8 CIccMan::CopyIccDeviceInfo(void*) + 44
2   com.canon.edsdk               	0x0000000108da7736 CMacIOManager::IsSupportedPtpCamera(void const*, char*, unsigned int*, int*, char*) + 292
3   com.canon.edsdk               	0x0000000108da7e4f CMacIOManager::GetPtpCameraPortInfo(SPortInfo*, unsigned int*) + 349
4   com.canon.edsdk               	0x0000000108da5b4d CIOManager::GetCameraPortInfo(SPortInfo*, unsigned int*) + 39
5   com.canon.edsdk               	0x0000000108dae7fb CEdsCameraList::EnumChildren() + 149
6   com.canon.edsdk               	0x0000000108d9f587 CEdsObject::GetChildCount(unsigned int*) + 101
7   com.canon.edsdk               	0x0000000108da089d CEdsdk::GetChildCount(__EdsObject*, unsigned int*) + 95
8   com.canon.edsdk               	0x0000000108d9b1e5 EdsGetChildCount + 65
9   info.v002.Camera-Live         	0x0000000108c8c9a5 0x108c8a000 + 10661
10  com.canon.edsdk               	0x0000000108da6ef4 CMacIOManager::HandleIccEvent(unsigned int, void*) + 72
11  com.apple.ImageCaptureCore    	0x00007fff37c29c4e -[ICMasterDeviceBrowser addDevice:] + 1364
12  com.apple.ImageCaptureCore    	0x00007fff37c29314 -[ICMasterDeviceBrowser updateDevices:type:] + 2561
13  com.apple.ImageCaptureCore    	0x00007fff37c286b0 -[ICMasterDeviceBrowser processDevices:] + 253
[... 405 lines cut ...]
CoreImage                           32K        4 
CoreUI image data                 1104K        9 
Foundation                           4K        1 
Kernel Alloc Once                    8K        1 
MALLOC                           183.5M       55 
MALLOC guard page                   32K        7 
MALLOC_NANO (reserved)           384.0M        1         reserved VM address space (unallocated)
Memory Tag 242                      12K        1 
STACK GUARD                       56.0M        8 
Stack                             11.6M        8 
VM_ALLOCATE                        116K       18 
__DATA                            23.9M      283 
__DATA_CONST                        24K        2 
__FONT_DATA                          4K        1 
__GLSLBUILTINS                    5176K        1 
__LINKEDIT                       354.6M       10 
__OBJC_RO                         32.0M        1 
__OBJC_RW                         1776K        1 
__TEXT                           170.5M      284 
__UNICODE                          564K        1 
mapped file                       57.7M       20 
shared memory                      644K       16 
===========                     =======  ======= 
TOTAL                              1.3G      766 
TOTAL, minus reserved VM space   903.0M      766

DMCTowns, 2020-04-01:
The app is excellent, by the way, so many thanks for your work on it.

bangnoise, 2020-04-01:
Thanks for the report - it looks like a resurfacing of an issue where Apple's ImageCapture framework conflicts with the Canon SDK.

I need to patch -[ICMasterDeviceBrowser processDevices:] to disable it.

You could try [alpha build 13](https://github.com/v002/v002-Camera-Live/releases/download/13/Camera.Live.zip) - it doesn't use the Canon SDK and may avoid the problem - but it's an alpha version with some major changes, so I can't promise it won't introduce any new problems of its own...

DMCTowns, 2020-04-01:
Thanks for your prompt response. I'll give alpha 13 a go and let you know if I come across any issues.

Ihatetomatoes, 2020-04-02:
I had similar issue when my **EOS 70D was disconnecting** after 20 or 30 seconds. My issue was that my USB cable was connected to USB hub and not directly to USB port on my iMac.

**Since connecting directly, I have no dropouts** and the connection works fine.

Hope this helps someone else others with similar issue.

DMCTowns, 2020-04-03:
> I had similar issue when my **EOS 70D was disconnecting** after 20 or 30 seconds. My issue was that my USB cable was connected to USB hub and not directly to USB port on my iMac.
>
> **Since connecting [...]

In my case I was running the camera directly into the Mac's USB port.

bangnoise, 2020-04-04:
@DMCTowns it would be good to know if you still hit this issue with alpha 13?

DMCTowns, 2020-04-06:
> @DMCTowns it would be good to know if you still hit this issue with alpha 13?

@bangnoise I did a live stream last night with 13 and it didn't crash, which was great. The app did display the 'Camera Error' notification but this didn't seem to affect the performance.

I need to do some more investigation, but 13 did seem to introduce a new problem: My set up is this:

```
Canon 80D -> Camera Live -\ 
                                                     - > OBS -> YouTube Live Stream
Zoom Window --------------/
```

With 11, this gave me a near faultless stream at 1080p/30ps. With 13, the Zoom output was very choppy and the camera feed had a strange 'lag & catch up' when mixing back from the Zoom to the camera.

Massive caveat: I'm running all this on a 2014 dual core i5 Macbook Pro, which simply may not be up to the task. I have a new i7 MBP arriving today, so I'll let you know if that cures it. But I wondered if 13 does take a lot more CPU time than 11?

bangnoise, 2020-04-07:
@DMCTowns the camera was working but the status said "Camera Error"?

The choppy/lagged output sounds like issue #92. It's not the laptop's fault.

DMCTowns, 2020-04-07:
> @DMCTowns the camera was working but the status said "Camera Error"?

@bangnoise, yes, that's correct.

DMCTowns, 2020-04-16:
@bangnoise , FYI, I've done a couple of live streams on a brand new MacBook, using Camera Live 11 and so far no issues.