v002/v002-Camera-Live #88: Are there expected problems going from High Sierra 10.13 to Mojave or Catalina?

ThSGM, 2020-04-04:
I am sorry if this is not a proper feature request, but I'm not sure where else to ask. Camera Live has been extremely useful in allowing me to deliver my lectures during this time. However, my university is forcing me to upgrade my macOS 10.13 to a newer OS, and I'm terrified that this will cause some issues with allowing me to continue delivering the lectures until the end of term when it can all be sorted properly.

I am using a Canon T3i with CamTwist and Camera Live to allow me to record in OBS. 

Looking at the issues forum, I see that some people did complain about issues with Catalina, and there were various releases a few versions ago that seemed to update it. How much should I be worried about this?

bangnoise, 2020-04-04:
I'd suggest trying the [latest alpha build](https://github.com/v002/v002-Camera-Live/releases/download/13/Camera.Live.zip) on your current macOS - if it works with your camera then it should continue to work on Catalina.

timhooker, 2020-04-15:
I can confirm that I have Catalina and a Canon EOS 70D. I used the alpha build and it works great.

Ender-54, 2020-04-30:
Are you using CamTwist with Camera Live? If so how are you getting around the Syphon disable? I'm trying to use this for zoom meetings etc, but CamTwist won't recognize ,y camera from Camera Live, I'm assuming it's due to the disabling of Syphon.