v002/v002-Camera-Live #89: Installation Instructions

kpu979, 2020-04-04:
Hi, When I download the ZIP file on my MAC, to a layman, it's a pile of gobbly gook.

How do I turn this code into an application that I can use?  In other words, Can you provide step by step installation instructions?

Believe me, I have searched the internet, and could not find anything useful.  My next step is to find an alternative, but it seems most articles point to this app as the way to turn my DSLR into a webcam.

bangnoise, 2020-04-04:
Which link did you follow? Either you have downloaded the source or your browser has failed to unzip the download properly. I have heard of Chrome having problems unzipping applications before.

[Release build 11](https://github.com/v002/v002-Camera-Live/releases/download/11/Camera.Live.zip)
[Alpha build 13](https://github.com/v002/v002-Camera-Live/releases/download/13/Camera.Live.zip)

closed 2020-04-04 by bangnoise

kpu979, 2020-04-04:
My mac (safari) unzipped what looks like the raw code (source), not in an application that I can use.  

Is there another way?  How does this get installed?