v002/v002-Camera-Live #90: Canon EOS M50 is active but can't be selected as webcam

midudev, 2020-04-05:
I'm able to get the `active` status in the CameraLive app but I'm not able to select the camera as device in any app used to record.

**System:**
 - macMini
 - macOS version: 10.15.4
 - Canon EOS M50
 - Camera Live version 11

**Describe the bug**
Camera is active but I can't select it in any app.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect Camara to PC by USB.
2. Open Camera Live.
3. Camera is detected and connected.
4. Select Camera. Camera Live gets "Active" status.
5. Try to use camera in any app. It's not detected.

**Expected behavior**
The new camera should be available to be selected in order to be used.

**Screenshots**
Camera is active:
[image]

But trying apps, I can't select the camera as webcam:
[image]

[image]

Camera is detected:
[image]

**Additional context**
If I try the latest alpha, the behaviour is exactly the same but the camera screen keeps in black. With the latest stable version I at least could see the camera working (but not detected in PC).

bangnoise, 2020-04-05:
Camera Live is not a virtual webcam driver. It provides a [Syphon](http://syphon.v002.info) output for other Syphon-aware applications to use. A popular way to use it as a camera is to pair it with CamTwist.

closed 2020-04-05 by bangnoise

midudev, 2020-04-05:
My bad. :) Thanks for your response and your awesome work with Camera Live. Much appreciated. 🙇

Jip-Hop, 2020-05-14:
Would be cool though if Camera Live could also act as a virtual webcam driver. The resolution coming out of the M50 is 1024x576. CamTwist doesn't support this resolution apparently. When CamTwist turns the Syphon feed into a virtual webcam, it either downscales or upscales this resolution. Upscaling introduces (upload) overhead when streaming. Also the new [obs-mac-virtualcam](https://github.com/johnboiles/obs-mac-virtualcam) driver doesn't currently create a webcam at the exact resolution of the M50. Additionally it would be great to just need 1 app to use the camera as a webcam, like [EOS Webcam Utility](https://www.usa.canon.com/internet/portal/us/home/support/self-help-center/eos-webcam-utility/) but for MacOS 🙂

vagari, 2020-05-15:
@Jip-Hop Do you know if the 1024x576 set in stone? I noticed the (well, what I assume was that) terrible quality when trying to use my M50 on Zoom. It was worse than the cheap built in webcam on my 2015 13".

Jip-Hop, 2020-05-17:
I think the quality is quite good actually, especially considering it's only 1024x576. I don't think it's possible to increase the resolution.

vagari, 2020-05-17:
@Jip-Hop That's odd. The quality looks good on my machine, but the video that my coworkers were seeing on Zoom was terrible. They could see I was using a nice setup, with a blurred background, but it was also super pixelated. I thought maybe what you mentioned could be the culprit. As I haven't found any rhyme or reason to it.

bangnoise, 2020-05-19:
@vagari sounds like a bandwidth problem with Zoom, nothing to do with Camera Live.

vagari, 2020-05-20:
Thanks @bangnoise, it's entirely possible. I thought perhaps Jip was onto something when he referenced the poor handling of the resolution (and the subsequent scaling). The M50 is the only device I have that's a proper camera. So I can't easily compare something else (using the same basic setup). Using other software with my iPhone wirelessly tethered worked fine in Zoom, as does the aforementioned built in webcam. Attached is the screenshot my coworker sent me while I was using the M50. Definitely _something_ weird going on. It's like just my face is being distorted. I was going to crop off coworkers, but instead I censored names, so everyone could see how clear they look on the same connection.
[image]

vagari, 2020-05-20:
Well that's interesting. I decided to try some other things... Opened up a meeting on my private Jitsi server. This is what my iPhone was seeing... 
[image] 

iPad was very similar. Then I opened up Screenflow and tried to pick both options listed for CamTwist (using Syphon and Camera Live and the only resolution option is 320x240. :-O 

I must be doing something wrong. Or have a bad USB cable or something. Assuming this is only me experiencing this. Any guidance welcome...

Jip-Hop, 2020-05-20:
You could try OBS with: https://github.com/johnboiles/obs-mac-virtualcam
Don't know what the actual issue is but can't hurt to try.

Also I've heard Zoom recently blocked virtual webcams. So you may not be able to use the M50 (or your iPhone for that matter) with Zoom any more.

Also make sure you put the M50 in video mode (not photo mode) and set it to 1080p.

bangnoise, 2020-05-20:
The virtual webcam driver (eg CamTwist) may be downscaling or upscaling to conform to conventional frame dimensions. You can see the resolution (and quality) coming out of Camera Live using any other Syphon tool - eg the test Simple Client ([download](https://github.com/Syphon/Simple/releases/download/5/Syphon.Simple.Apps.zip)).

vagari, 2020-05-20:
Wow, anyone have an alternative to CamTwist? I think that's the culprit... @bangnoise that Syphon tool worked great. It showed that the feed was in fact the proper resolution. @Jip-Hop I also verified I was at least getting higher quality with OBS (unless set to "high" it maxed out at 960xsomething to start). I then went into CamTwist and found that, though I hadn't adjusted any settings, it was defaulting to 320x240 (well there's your problem). 

I've tried adjusting it to the max it will let me, or even setting it to the custom actual resolution of the feed, and that got things a little better (at least it wasn't 320x240). But when trying to use Jitsi again, it worked in a Chrome based browser, but wouldn't in Firefox. And [this WebRTC resolution test site](https://webrtchacks.github.io/WebRTC-Camera-Resolution/) is all over the map in regards to what it will handle, and it fluctuates. I also get vastly different results when using "CamTwist" vs "CamTwist (2VUY)". With the latter getting up to about 640x480 (yay, twice the resolution *facepalm*). [This one](https://webrtc.github.io/samples/src/content/getusermedia/resolution/) that only uses more standard sizes, doesn't work at all (in Firefox). 

Thanks again for helping me troubleshoot. I'm gonna go try and find something other than CamTwist, which doesn't seem to like the M50. And as mentioned, if anyone has another option I'm all ears.

Jip-Hop, 2020-05-20:
Hi Jake, I already mentioned an alternative to CamTwist in my previous comment.
Try OBS with https://github.com/johnboiles/obs-mac-virtualcam <https://github.com/johnboiles/obs-mac-virtualcam>.
And check out this issue: https://github.com/johnboiles/obs-mac-virtualcam/issues/83 <https://github.com/johnboiles/obs-mac-virtualcam/issues/83>.
My issue with it, is that the output is upscaled.
But that's better than downscaled ;)

vagari, 2020-05-20:
Oh, I see, you mean to use OBS with Zoom/Jitsi, etc. I tried playing with the camera, I didn't try to use it as a passthrough to other things. I'll take a look at that after I actually get some work done today. lol 

Oh, and the issue with Zoom blocking virtual cams. I found a couple articles, [like this one](https://medium.com/@dkiesow/how-to-fix-the-virtual-camera-function-in-zoom-mac-only-9872a6d9ea5), where if you have Xcode (which I already did) you can reset things and the next time you launch Zoom it'll ask to have access to the camera/microphone again and voila, the virtual cams are back. It has to be done after each update and likely won't work forever. But so far it's at least allowed me to use EpocCam with my phone while trying to sort out the issues with the M50.

vagari, 2020-05-20:
Thanks again @Jip-Hop (and @bangnoise). I successfully got the camera working with OBS and did my daily Zoom call with much better quality. =)

Jip-Hop, 2020-05-20:
Glad you got it working :)

Jip-Hop, 2020-05-28:
Just released by Canon: [EOS Webcam Utility Beta for MacOS](https://community.usa.canon.com/t5/Software/Now-Available-for-Mac-EOS-Webcam-Utility-Beta-Software-Share/m-p/307778). Works with my Canon EOS M50 @ 1024x576 resolution.