v002/v002-Camera-Live #92: Camera Live 13 - Stuttering/Skipping frames

tavon, 2020-04-06:
**PLEASE COMPLETE ALL OF THIS SECTION**

**System:**
macOS version: 10.15.3
Camera manufacturer and model: Canon EOS 70D
Camera Live version: 13

**Describe the bug**
- Camera Live 13 causes the footage from the camera to be choppy. It looks like it's stuttering and the footage is not smooth.
- Camera Live 11 works great.

**To Reproduce**
Steps to reproduce the behavior:
- Install Camera Live 13 and open any application that reads a webcam

**Expected behavior**
- Smooth playback

**Screenshots**
Camera Live 13
https://www.dropbox.com/s/5u3ev1yvt4b4bcv/camera%20live%2013.mp4?dl=0

Camera Live 11
https://www.dropbox.com/s/2vbxf55nbra7zzg/Camera%20Live%2011.mp4?dl=0

bangnoise, 2020-04-07:
May be related to #75 though that issue affected 11 too.

The next build will have an attempt to improve this, and more logging to get to the bottom of the problem.

Even though it works well in 11 it may be worth noting that a lot of issues like this are down to problems with USB - try different cables, don't route through a hub, unplug other devices.

tcchau, 2020-05-22:
@bangnoise First, many thanks and kudos for developing a great and very useful utility. I'm commenting here because I experience this problem in versions 12 and 13, but not 11.

My set up is 
 - Canon 70D, directly into 
 - Macbook Pro 13" 2017, without USB hub

Other notes
 - USB cable is new, directly from mini-USB to USB-C
 - The stuttering can be observed in the Live View when Camera Live is running, without any other Syphon clients like OBS or CamTwist
 - On my 70D at least, when Camera Live is running, it also causes the Quick Menu on the touch screen of the camera to stop working--the menu overlay comes on for a split second and then disappears. Maybe that's a clue as to what's happening
 - The stuttering does not happen when using Canon EOS Utility for Remote Shooting, even in video mode