v002/v002-Camera-Live #93: Canon EOS 750D not working

Operaiter, 2020-04-07:
**PLEASE COMPLETE ALL OF THIS SECTION**

**System:**
 - macOS version: 10.15.2
 - Camera manufacturer and model Canon EOS 750D
 - Camera Live version 11 and 13(beta)

**Describe the bug**
Camera Live is not showing Canon EOS 750D.
The second test camera Canon EOS 550D is working finde

**To Reproduce**
Steps to reproduce the behavior:
1. Start Software
2. Connect Camera

**Expected behavior**
Camera should be shown in list

dantrauner, 2020-04-14:
I'm wondering if I'm having the same issue – on 10.15.3 with a T2i but the camera just acts like it's not connected at all. Should I be setting it to a certain mode before it's expected to be picked up via USB?

juriroemer, 2020-04-29:
I had the same issue with a 750D - resolved by deactivating Wi-Fi/NFC.

cheers