v002/v002-Camera-Live #95: Camera Live gets stuck at Starting.

ddwwcc, 2020-04-07:
**PLEASE COMPLETE ALL OF THIS SECTION**

**System:**
 - macOS version: 10.14.6 
 - Camera manufacturer and model: Canon EOS 7D mk i & EOS 5D mk iii
 - Camera Live version: 11

**Describe the bug**
When reconnecting a camera, the connection report does not go beyond Starting

**To Reproduce**
Steps to reproduce the behavior:
1. Connect camera via USB
2. Open Camera Live, 
3. Select camera. 
4. Press Start/ Stop button on camera (Note this is only required on 7D, not 5D)
5. Video feed shown as active.
6. Quit Camera Live
7. Open Camera Live
8. Camera is already selected, and shown as Starting, but never gets to Active.

**Expected behaviour**
At 7. Above, camera should reconnect, and be shown as Active.

**Screenshots**
If applicable, add screenshots to help explain your problem.

Naaa

**Additional context**
Add any other context about the problem here.

Have found a reliable way of reconnecting, which may show where the problem lies.

From above sequence, this reliably reconnects.

1. Connect camera via USB
2. Open Camera Live, 
3. Select camera. 
3. Press Start/ Stop button on camera (Note this is only required on 7D, not 5D)
4. Video feed shown as active.
5. De-select camera by clicking on camera live window, below the listed camera.
6. Quit Camera Live
7. Open Camera Live 
8. Select camera. 
9. Press Start/ Stop button on camera (Note this is only required on 7D, not 5D)
10. Video feed shown as active.

So, it looks like the problem is related to the camera already being selected, when Camera Live is re-opened.

ljakab, 2020-04-13:
I have the exact same issue with a 5M2, thanks for posting the workaround! Would be nice if the bug could be fixed though...

djruess, 2020-04-17:
This worked for the Canon 700D/5ti as well, except there isn't a start and stop button to press. I simply deselected the camera from Camera Live and then quit Camera Live, Reopened Camera Live, and selected the camera, and it worked. Thank you!